dawidchyrzynski/arduino-home-assistant #2: availability topic in MQTT

tomakara, 2021-01-31:
Hey there!
Thanks for nice library!
Could you please add possibility for setting  availability_topic in MQTT for devices? For example if a switch or sensor is temporary not working.

dawidchyrzynski, 2021-02-02:
Hi! 
Sure, I will add the requested feature by the end of this week.

dawidchyrzynski, 2021-02-06:
Hi @tomakara
I have already released a new version of the lib (1.1.0). Feel free to test the availability feature. If you will find any issue please open a new ticket.

closed 2021-02-06 by dawidchyrzynski

tomakara, 2021-02-06:
Thanks man, that was a lot of work. I'm just a script kiddie but it's a nice evolution in the code. Going to try it now.

dawidchyrzynski, 2021-02-06:
I had to refactor and simplify a few things in the codebase. Introducing new features in the future will be simpler for sure.

tomakara, 2021-02-06:
Works like a charm!