dawidchyrzynski/arduino-home-assistant #3: Wifi integration

Sichroteph, 2021-02-03:
Thank you for this very interesting code.
Do you know a way to change the code so the board can contact home assistant with the wifi network instead of ethernet ?

dawidchyrzynski, 2021-02-03:
Hi!
There is an example showing how to integrate the library with NodeMCU that is equipped with a Wi-Fi chip. At this moment that's the only way to use Wi-Fi. Is this solution satisfying for you?

https://github.com/dawidchyrzynski/arduino-home-assistant/blob/main/examples/nodemcu/nodemcu.ino

Sichroteph, 2021-02-03:
I will dig from there, thanks. I have a arduino nano 33 iot but the code should remain essentialy the same.

Sichroteph, 2021-02-04:
It is recommended to use the WiFiNINA library with the arduino nano 33 IOT.

However, the WiFiNINA  WiFiClient seems to be different that the ESP8266WiFi since this code  

`#include <WiFiNINA.h>
#include <ArduinoHA.h>

byte mac[6];
WiFiClient client;

HADevice device(mac, sizeof(mac));
HAMqtt mqtt(client, device);`

raise this error : (I paste only the beginning)
_In file included from C:\Users\Christophe\Documents\Arduino\libraries\home-assistant-integration\src/ArduinoHA.h:5:0,
                 from F:\Mes documents\Programmation\Codage\Arduino\Arduino nano 33 IOT\tests MQTT\sketch_feb04a\sketch_feb04a.ino:2:
C:\Users\Christophe\Documents\Arduino\libraries\home-assistant-integration\src/HAMqtt.h:17:18: error: expected ')' before '&' token
     HAMqtt(Client& netClient, HADevice& device);
                  ^
C:\Users\Christophe\Documents\Arduino\libraries\home-assistant-integration\src/HAMqtt.h:18:34: error: reference to 'Client' is ambiguous
     HAMqtt(const char* clientId, Client& netClient, HADevice& device);
                                  ^~~~~~
C:\Users\Christophe\Documents\Arduino\libraries\home-assistant-integration\src/HAMqtt.h:7:7: note: candidates are: class Client
 class Client;
       ^~~~~~_

dawidchyrzynski, 2021-02-04:
I will get back to you soon. I ordered nano 33 IOT to test it out on my own.

dawidchyrzynski, 2021-02-07:
Hi @Sichroteph 

I have managed to fix all issues related to Arduino Nano 33 IoT. Feel free to use the library in your project. You can find the example code here: https://github.com/dawidchyrzynski/arduino-home-assistant/blob/main/examples/nano33iot/nano33iot.ino

closed 2021-02-07 by dawidchyrzynski

Sichroteph, 2021-02-07:
Thank you so much. Merging your work to this project now opens very exciting opportunities in home automation 👍
https://www.okdo.com/project/bluetooth-triggered-xmas-lights/

Sichroteph, 2021-02-08:
Your nano 33 IOT update works fine.
- Is there a  sort of  mqtt.disconnect() method to properly cut off the mqtt connection ?

My issue is that BLE and Wifi doesn't seem to properly work together (I did not succeed in detecting BLE devices when Wifi is on)
So my script now activate the wifi or the BLE depending of my need. But doing this way makes the arduino be kicked by the mqtt broker for timeout reasons.

dawidchyrzynski, 2021-02-09:
There is no way to disconnect from the MQTT broker right now. I will implement this feature today.

Did you try to just disconnect from the Wi-Fi AP? HAMqtt should switch into a disconnected state when the connection is lost.

Sichroteph, 2021-02-09:
I tried indeed, my code triggers :

WiFi.disconnect();
WiFi.end(); 

After the sensor.setState method. 
Doing so doesn't close the connection with the broker :

1612825488: New client connected from 192.168.1.150 as 34d53abd9e7c (p2, c1, k15, u'totofaki').
1612825510: Client 34d53abd9e7c has exceeded timeout, disconnecting.

Right now my only theory is that beeing kicked out prevent a further connection. 
The best thing from me would be to be able to poll BLE and mqtt at the same time. Still working on it.

Sichroteph, 2021-02-09:
I tried with a different method,  BLE.scan() instead of BLE.poll() but the issue remains. Not sure if it's due to your integration, I will try to use another MQTT one.

Here is the code :
https://pastebin.com/yJgR6Q34

The BLE and MQTT part are working very well when running independently. 
However when combined mqtt.loop(); freeze the device.

Sichroteph, 2021-02-09:
I know now for sure that your integration is not in fault here.
I managed to publish MQTT packets with PubSubClient.h, and again it is working fine until BLE is turned on.

The code is here :
https://pastebin.com/W1QnCUwH

So it seems to be a hardware limit of some kind. I will wire the nano 33 iot in BLE mode to a weimos d1 mini that will trigger home assistant. That's a bit twisted to do so but I see no other way to do.

dawidchyrzynski, 2021-02-09:
According to [this thread](https://forum.arduino.cc/index.php?topic=706766.0) it seems like it's not possible to use BLE and WiFi at the same time.

Sichroteph, 2021-02-09:
Thank you that is clear now !

hanen1234, 2021-03-11:
Who knows how installed library ds 1307 in arduino

dawidchyrzynski, 2021-03-11:
@hanen1234 do you mean RTC DS1307? What's your goal?

hanen1234, 2021-03-11:
Yes RTC DS1307

dawidchyrzynski, 2021-03-11:
Integrating the RTC library shouldn't be any problem in your project as ArduinoHA is just a communication layer for Home Assistant. What do you exactly want to achieve?

hanen1234, 2021-03-11:
In the end I want the program arduino to appear to me in the last article
showing me time and temperature.

hanen1234, 2021-03-11:
So i need RTC DS1307