dawidchyrzynski/arduino-home-assistant #6: MQTT credential

Sonusss, 2021-02-09:
Hi,

I'm trying MQTT credential specified as per your example
#define BROKER_USERNAME     "user" // replace with your credentials
#define BROKER_PASSWORD     "pass"

But seems not working this is what I get in MQTT server log:

1612873030: New connection from 192.168.1.144 on port 1883.
1612873030: Socket error on client <unknown>, disconnecting.

tomakara, 2021-02-09:
Hi Sonusss, that's not much info you provide. Have you tried to search some solution? I recommend for example this thread: https://stackoverflow.com/questions/35704632/mqtt-socket-error-on-client-unknown 
Let us know if something there helped you.
T.

dawidchyrzynski, 2021-02-09:
Hi @Sonusss,

Please check your credentials. Most likely they're invalid.

You can also turn on debug mode in the library by uncommenting #define in [this file](https://github.com/dawidchyrzynski/arduino-home-assistant/blob/main/src/ArduinoHADefines.h). In order to get logs working you need to setup serial at the beginning of the setup procedure (`Serial.begin(9600);`);

Sonusss, 2021-02-09:
Solved !

Shame on me...
Stupid mistake, I forgot to pass credential parameters with mqtt.begin

closed 2021-02-09 by Sonusss