dawidchyrzynski/arduino-home-assistant #7: ADD: Last Will and Testament (LWT)

LKuech, 2021-02-24:
Hi,

great library. Very good work!

Is it possible, that you add the LWT function of MQTT? This way the broker will automatically set the availability back to "offline" if the device is active for some time (like Tasmota devices are able to do it).

dawidchyrzynski, 2021-02-25:
Hi @LKuech 

I'm not entirely sure if it will be possible. Tasmota uses its own integration on the Home Assistant side. I don't see anything related to the device's availability in the built-in MQTT integration. I will research this and get back to you.

LKuech, 2021-02-25:
Hi.

thanks for your response.
Take a look at the "enableLastWillMessage()" function of the EspMQTTClient. They are doing this as well.
[https://github.com/plapointe6/EspMQTTClient](url)

Also this seems a good explanation:
[https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/](url)

dawidchyrzynski, 2021-03-21:
Hi @LKuech 

This feature is already available on the `develop` branch. Please let me know if all works as expected.

closed 2021-03-23 by dawidchyrzynski