dawidchyrzynski/arduino-home-assistant #12: Icon type

raducatanet, 2021-03-08:
Hello Dawid!

Nice job with this libraries.
Can you please add support for "icon type" in actual sensors definitions.

Thank you!

raducatanet, 2021-03-19:
Thank you, Dawid!

dawidchyrzynski, 2021-03-19:
No problem. Please note that this feature is currently available on the `develop` branch. You can test the branch before it will be released as 1.2.0.

raducatanet, 2021-03-19:
I'll tested in weekend and provide you a feedback. Have a nice day!

raducatanet, 2021-03-19:
Hello Dawid, 

I use e ESP8266 (D1 mini module) as a WiFi module that communicate on GPIO_02 as TX pine to communicate with a Mega 2560.
I'm not a SW expert, but with the new library, the serial communication over GPIO_02 is down and the ESP is resetting. I put the stack in Exception Decoder and here is the result:

Decoding stack results
0x4021b548: HardwareSerial::write(unsigned char const*, unsigned int) at D:\Apps\Arduino_IDE\arduino-1.8.13\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 165
0x4021b53c: HardwareSerial::write(unsigned char const*, unsigned int) at D:\Apps\Arduino_IDE\arduino-1.8.13\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 164
0x40219f1f: DeviceTypeSerializer::mqttPublishMessage(BaseDeviceType const*, char const*, char const*) at d:\Apps\Arduino_IDE\arduino-1.8.13\portable\sketchbook\libraries\arduino_28289\src\device-types\DeviceTypeSerializer.cpp line 312
0x4020ebd3: connectasclient(unsigned short, bool, bool) at C:\Users\uidp6264\AppData\Local\Temp\arduino_build_901907\sketch/wifi_manager.h line 107
0x4010045c: millis() at D:\Apps\Arduino_IDE\arduino-1.8.13\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 188
0x4021a04a: HASwitch::publishState(bool) at d:\Apps\Arduino_IDE\arduino-1.8.13\portable\sketchbook\libraries\arduino_28289\src\device-types\HASwitch.cpp line 136
0x4021a079: HASwitch::setState(bool) at d:\Apps\Arduino_IDE\arduino-1.8.13\portable\sketchbook\libraries\arduino_28289\src\device-types\HASwitch.cpp line 73
0x4020853a: check_heat_required() at C:\Users\uidp6264\AppData\Local\Temp\arduino_build_901907\sketch/task_os.h line 261
0x4020f35c: one_scond_task(bool) at C:\Users\uidp6264\AppData\Local\Temp\arduino_build_901907\sketch/task_os.h line 157
0x4020f82f: time_tasks(bool) at C:\Users\uidp6264\AppData\Local\Temp\arduino_build_901907\sketch/task_os.h line 50
0x4020f863: loop() at D:\Apps\Arduino_IDE\arduino-1.8.13\portable\sketchbook\Mega2560\Pellets_burner\SW\esp8266_wifi_interface_V2/esp8266_wifi_interface_V2.ino line 50
0x4021e140: loop_wrapper() at D:\Apps\Arduino_IDE\arduino-1.8.13\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 197

If I comment the setState calls in my SW, everything is working perfect.  I don't have this problem with the library from the main branch.

Regarding the setIcon implementation is working like a charmed.

[image]

Since the issue here is not related to setIcon, I believe the topic can be closed.

Thank you and have a nice weekend!

raducatanet, 2021-03-20:
Hello again! :)

There is no problem with the library,  it seems to be a power supply issue. 
I change a capacitor on the power line and now everything is working perfectly.

Thank you!

closed 2021-03-22 by raducatanet