dawidchyrzynski/arduino-home-assistant #13: Text message as payload

raducatanet, 2021-03-08:
Hello Dawid!

Can you please add a sensor that transmit text message in payload. 
I find very useful to update a status of a multi-switch to display text message instead of number.
For example if outside is "sunny", "cloudy" or "rainy", or for example in errors messages. Is helpful to have a error element that display the error in text format instead of number.

Thank you in advance!

dawidchyrzynski, 2021-03-20:
Hi @raducatanet 

The feature is available on the develop branch. Please note that I have refactored HASensor that is not compatible with the previous implementation (see: https://github.com/dawidchyrzynski/arduino-home-assistant/blob/develop/examples/sensor/sensor.ino).

raducatanet, 2021-03-21:
Hello Dawid!

I hope my wife is not going to kill me.
Grate job with this library. 

I have tested also the text message as payload and is working perfect.
I'll implement the error messages in text , maybe next weekend, but for my System State is as I wanted to be.
You can see the red square pointing the pellets burner system state.
[image]
Regarding this topic, I'll come back with some feedback after implementing the ERROR message as text and we can close the topic after that. 
Thank you very much for your time and support!

I have also test a little bit the HVAC, great job also. I hope you don't mind writing a few words about.
I did not figure out yet how to change the Off status from the blue square, still on working :) . 
In the config panel it say Off (Auto), and I do not understand what is Off, but maybe I have to dig dipper.
[image]
Any way, grate job.

Thank you and have a nice Sunday!

dawidchyrzynski, 2021-03-21:
Hi @raducatanet 

Thank you for the details! I really appreciate your effort in improving this library.

OFF label in the panel is displayed due to "action" property of the HVAC. Home Assistant's documentation doesn't clearly state what's the difference between "action" and "mode". The "action" may be only produced by the device (there is no command for changing it). I suppose this was created for thermostats that have physical buttons for HVAC operations, so in such case "action" is used instead of "mode".

You correctly noticed that in the current approach this doesn't make sense so I decided to add a new HVAC feature called "ActionFeature". You may implement your HVAC in two different ways:
1) don't use "ActionFeatue" in the HAHVAC constructor and rely on modes. I suppose that's the thing you want to achieve. In this way, you will be able to change HVAC's modes from the panel.
2) use "ActionFeature" and call "setModes(0)" method. This will result in hiding modes in the HA panel. Please note that in this case, your device will need to publish actions (using "setAction" method) based on some user's interaction (e.g. buttons on the device). In this approach, you won't be able to change actions from the HA.

raducatanet, 2021-03-22:
Hi @dawidchyrzynski 

I'm using the first method, with no "ActionFeatue" in the HAHVAC constructor and rely on modes.
I managed twice to make it work as I wanted, and the Off was changing to "Heat" when mode was changed, but most of the time is still display "Off" even if the mode is "Heat".
[image]
Here are my hvac lines:
//*************************************************************************

HAHVAC hvac("House_Thermostat");

void onTargetTemperatureChanged(double temp) { // hvac
  if (new_targ_temp == false) // params were not updated via html
  {
    Serial.print("Target temperature: ");
    Serial.println(temp);
  }
}

void onModeChanged(HAHVAC::Mode mode) { // hvac mode change
  Serial.print("Mode: ");
  if (mode == HAHVAC::OffMode)
  {
    Serial.println("Off"); 
    DISABLE_HEATING = YES;
  }
  else if (mode == HAHVAC::HeatMode)
  {
    Serial.println("Heat"); 
    DISABLE_HEATING = NO;
  }
}

void updateHvacState(){ //update hvac state if changes are done in the web interface ## this function is called every 1 second
  static boolean old_DE = NO;

  if (old_DE == DISABLE_HEATING)
    return;

  if (DISABLE_HEATING == YES)
      hvac.setMode(HAHVAC::OffMode);
  else
      hvac.setMode(HAHVAC::HeatMode);
  old_DE = DISABLE_HEATING;
}

// here are the config lines from setup

  // configure HVAC actions
  hvac.setModes(HAHVAC::HeatMode + HAHVAC::OffMode); 
  hvac.onModeChanged(onModeChanged);
  hvac.onTargetTemperatureChanged(onTargetTemperatureChanged);

  // configure HVAC (optional)
  hvac.setName("Home_Thermostat");
  hvac.setMinTemp(16);
  hvac.setMaxTemp(28);
  hvac.setTempStep(0.5);
  hvac.setTargetTemperature(22.0);
  hvac.setRetain(true);
//****************************************************************
and this is about all for the hvac;
If you have any suggestion, please advize!

Thank you!

PS: I use a ESP8266 with:

Serial ports initialised with baud rate: 115200
Core Version: 	2_7_4
Sdk Version: 	2.2.2-dev(38a443e)
Cpu Frequency: 	160 MHz

dawidchyrzynski, 2021-03-22:
I'm not able to reproduce the problem. The only thing that comes to my mind is connection problem due to quality (e.g. long range connection with access point). Can you try this example:
```cpp
#include <ESP8266WiFi.h>
#include <ArduinoHA.h>

#define BROKER_ADDR     IPAddress(192,168,0,28)
#define WIFI_SSID       "YourSSID"
#define WIFI_PASSWORD   "YourPassword"

WiFiClient client;
HADevice device;
HAMqtt mqtt(client, device);

// see src/device-types/HAHVAC.h header for more details
HAHVAC hvac("Home_Thermostat");
unsigned long lastTempPublishAt = 0;
double lastTemp = 0;

void onTargetTemperatureChanged(double temp) {
    Serial.print("Target temperature: ");
    Serial.println(temp);
}

void onModeChanged(HAHVAC::Mode mode) {
    Serial.print("Mode: ");
    if (mode == HAHVAC::OffMode) {
        Serial.println("off");
    } else if (mode == HAHVAC::AutoMode) {
        Serial.println("auto");
    } else if (mode == HAHVAC::CoolMode) {
        Serial.println("cool");
    } else if (mode == HAHVAC::HeatMode) {
        Serial.println("heat");
    } else if (mode == HAHVAC::DryMode) {
        Serial.println("dry");
    } else if (mode == HAHVAC::FanOnlyMode) {
        Serial.println("fan only");
    }
}

void setup() {
    Serial.begin(9600);
    Serial.println("Starting...");

    // Unique ID must be set!
    byte mac[WL_MAC_ADDR_LENGTH];
    WiFi.macAddress(mac);
    device.setUniqueId(mac, sizeof(mac));

    // connect to wifi
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500); // waiting for the connection
    }
    Serial.println();
    Serial.println("Connected to the network");

    // assign callbacks (optional)
    hvac.setModes(HAHVAC::HeatMode + HAHVAC::OffMode);
    hvac.onTargetTemperatureChanged(onTargetTemperatureChanged);
    hvac.onModeChanged(onModeChanged);

    // configure HVAC (optional)
    hvac.setName("Home_Thermostat");
    hvac.setMinTemp(16);
    hvac.setMaxTemp(28);
    hvac.setTempStep(0.5);
    hvac.setCurrentTemperature(22.0);
    hvac.setRetain(true);

    mqtt.begin(BROKER_ADDR);
}

void loop() {
    mqtt.loop();

    if ((millis() - lastTempPublishAt) > 3000) {
        hvac.setCurrentTemperature(lastTemp);
        lastTempPublishAt = millis();
        lastTemp += 0.5;
    }
}
```

I tested it for a while on NodeMCU and ESP-01 and all works perfectly fine.

dawidchyrzynski, 2021-03-22:
You can also enable debug mode of the library by uncommenting `#define ARDUINOHA_DEBUG` in the https://github.com/dawidchyrzynski/arduino-home-assistant/blob/main/src/ArduinoHADefines.h file. It will tell us if connection with MQTT broker is fine.

closed 2021-03-23 by dawidchyrzynski

raducatanet, 2021-03-23:
Hello @dawidchyrzynski,

thank you for your support and work.
I have try the SW from your post. The only difference is the "mqtt.begin(BROKER_ADDR, mqtt_user, mqtt_password);".
I also activate the debug. All seems to be OK:

Initializing ArduinoHA
Server address: 192.168.123.44:1883
Connecting to the MQTT broker... Client ID: 840d8e8c8407
Connected to the broker
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/config, payload length: 600
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/at, payload length: 3
Subscribing topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ttct
Subscribing topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/mct
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 4
Received message on topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ttct, payload length: 4
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ttst, payload length: 5
Target temperature: 22.00
Received message on topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/mct, payload length: 4
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/mst, payload length: 4
Mode: heat
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 4
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 4
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 4
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 5
Received message on topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/mct, payload length: 4
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/mst, payload length: 4
Mode: heat
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 5
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 5
Received message on topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/mct, payload length: 3
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/mst, payload length: 3
Mode: off
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 5
Received message on topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/mct, payload length: 4
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/mst, payload length: 4
Mode: heat
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 5
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 5
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 5
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 5
Publishing message with topic: homeassistant/climate/840d8e8c8407/Home_Thermostat/ctt, payload length: 5

The result is the same: 
[image]

Maybe is a server config, or something on the HA server. Don't bother, is something I can live with :).

Thank you again and have a nice evening!