dawidchyrzynski/arduino-home-assistant #20: Problem with Home assistant Auroconfiguration

Corrado-Cuneo, 2021-03-22:
I use home assistant from many year.. I test your beatiful library with an Arduino Uno + Ethernet Shield... (I set IP manually and the shield connect ok to network and have correct ping response.
I'use your simple example "binary-sensor" schetch but I've no notification on home assistant of new device detected....
Any suggestion?
Is necessary additional task for connection?
Thanks!!
Corrado

dawidchyrzynski, 2021-03-22:
Hi @Corrado-Cuneo

Please confirm if your device is successfully connecting to the MQTT broker. You can enable debug mode of the library by initializing serial port (`Serial.begin(9600);`) in your `void setup() { }` and by uncommenting `#define ARDUINOHA_DEBUG` in the https://github.com/dawidchyrzynski/arduino-home-assistant/blob/main/src/ArduinoHADefines.h file.

Corrado-Cuneo, 2021-03-22:
Thanks!!! I've enable debug...
and I see message like this "Broker communication error"
In the Mosquito broker configuration I try to set:
anonymous: true (default is false)

And the log change to:
Connected to the broker
Publishing message with topic: homeassistant/binary_sensor/0010fa6e384a/input/config, payload length: 187
Publishing message with topic: homeassistant/binary_sensor/0010fa6e384a/input/state, payload length: 2
In HA no new device....

In my schetch I've set the Broker credential...

#define BROKER_USERNAME     "corrado" // replace with your credentials
#define BROKER_PASSWORD     "Xtgrbk2020IoT" //this is only a test password...

With this credential I've many tasmota device that work fine.
_

dawidchyrzynski, 2021-03-23:
Hi @Corrado-Cuneo

Which version of MQTT integration do you use?

Please double-check if you have this option turned on:
[image]

You can also check if Home Assistant is able to track MQTT messages by using "Listen to a topic" feature from the MQTT integration. Type `homeassistant/binary_sensor/0010fa6e384a/input/config` topic, click "Start listening" and then reboot your device (binary sensor).
[image]

Corrado-Cuneo, 2021-03-23:
Hi!!
I've check you suggestion:
[image]

[image]

[image]

If I set Anonimous=true

[image]
But no new entity discovered by home assistant...

If I set Anonimous false
[image]

[image]

dawidchyrzynski, 2021-03-26:
It looks like a problem with the network or mosquito, but I'm not 100% sure. Some people reported similar behavior in this here: https://github.com/eclipse/mosquitto/issues/523

Here are some ideas that may help:
* Use a different Ethernet shield (I mean the same model but a different device). The problem may be related to hardware/client ID
* Set the static IP address in `void setup()` instead of relying on DHCP
* If you have some spare router connect your PC with MQTT broker and Arduino to it and check if it solves the problem

I will do some extra research during the weekend. Please get back to me if the above solutions helped you.

Corrado-Cuneo, 2021-03-26:
I've install a new instance of Homeassistant on RPI....
I've set static IP on my sketch...
I load the new version of the library (1.20)...
Binary sensor OK!!!
In the weekend I test the other sketch and mixed sensor and switch...
Thanks for all!!! Usefull library!!

closed 2021-03-27 by dawidchyrzynski