dawidchyrzynski/arduino-home-assistant #22: Different prefix for discovery and sensor data

netixx, 2021-03-25:
Hello,

Most of the mqtt implementation I use (namely esphome) use a different prefix for the discovery topic (usually `homassistant`) and sending the data (usually, the name of the device).

I have a project in which I cannot use esphome (4G/LTE uplink, which is not supported yet), and I found that this lib covers a lot of my basis.

Is there any way to configure a different prefix for discovery and for sending the data ?

dawidchyrzynski, 2021-03-26:
I am not entirely familiar with this concept. At this moment it's not supported. Can you explain why topics related to sending data don't use `homeassistant` prefix? It would be also nice if you can provide me more details about your use case so I can get a better understanding of your needs. From my perspective, you don't need to care about topics as they are generated automatically for all devices' types and provided to HA through `config`. HA automatically subscribes to all of them so it doesn't matter what topic name you use internally.

netixx, 2021-03-26:
I don't know exactly the reason why projects like esphome do this. I guess the goal is to have a different topic for signalization and actual data.

The reason why I discovered this, is that I am running an mqtt configuration with a bridge between two brokers. At the moment, I am using mapping of prefix. In the config payload, the library is sending the prefix in the json payload, however, because the prefix is remapped, homeassistant gets wrong information.

I guess we would need to add a function to set the data prefix (as opposed to discovery prefix), by default, the data prefix would be the name/id of the device.

dawidchyrzynski, 2021-03-29:
I still don't see any use case for having two different prefixes. The current implementation uses different topics for each operation. The structure of topics is `[PREFIX]/[DEVICE TYPE]/[DEVICE ID]/[UNIQUE_ID/[TOPIC]`. Example topics produced by the library:
- `homeassistant/sensor/0010fa55444a/temperature/config` - configuration of HASensor is published under this topic
- `homeassistant/sensor/0010fa55444a/temperature/value` - value updates of HASensor are published under this topic
- `homeassistant/sensor/0010fa55444a/temperature/avail` - availability of the HASensor (if configured)

If you would like to replace discovery prefix with data prefix, the result will be:
- `homeassistant/sensor/0010fa55444a/temperature/config`
- `[YOUR PREFIX]/sensor/0010fa55444a/temperature/value`
- `[YOUR PREFIX]/sensor/0010fa55444a/temperature/avail`

If you're using a bridge between two brokers just configure it to pass through `homeassistant/*` topics.

dawidchyrzynski, 2021-03-29:
What MQTT bridge do you use?

dawidchyrzynski, 2021-03-29:
Ok, I think I see what you're going to achieve. If the above example with two different prefixes is fine then I can implement this feature. Please note that the "data" prefix" will be applied to commands and availability topics as well. Is that fine?

netixx, 2021-03-30:
I agree with you, this is not required to function properly, I could bridge the "homeassistant" topic without remapping, and it would work.

What you suggest is exactly what other libs do, and would be great. To retain retro-compatibility, `[YOUR PREFIX]` could default to the current `discovery_prefix`.

My suggestion was more to allow a different prefix to have the same behavior as other homeassistant/MQTT projets.

dawidchyrzynski, 2021-04-12:
Hi @netixx 

This feature is implemented and will be released soon. You can check the example here: https://github.com/dawidchyrzynski/arduino-home-assistant/blob/develop/examples/mqtt-advanced/mqtt-advanced.ino

closed 2021-04-17 by dawidchyrzynski