dawidchyrzynski/arduino-home-assistant #24: Send analog control fron HA to Arduino

Corrado-Cuneo, 2021-03-29:
Hello!!
I need to send a value from HA to Arduino... like a speed regulation or light regulation, a variable in a range (like a regulation...)
I need something like your fan speed regulation but with continue value in a range...
Is there something that can I use?
Thanks to all!!
Corrado

dawidchyrzynski, 2021-04-01:
Hi @Corrado-Cuneo,

Seems that Home Assistant is able to generate custom MQTT messages. I will research this and get back to you soon.

Corrado-Cuneo, 2021-04-04:
Thanks!!!
I think that a similar "fans control" (with possibility to set value and not only low-medium-hight)... or light control can solve my issue...
I need to save the date in arduino var and use in my schetc
Excuseme for 
bad english!!

dawidchyrzynski, 2021-04-12:
Hi @Corrado-Cuneo 

I have made a couple of changes that will allow you to handle custom MQTT messages. They are available on the `develop` branch and will be released soon. You can check the subscription&publishing example here: https://github.com/dawidchyrzynski/arduino-home-assistant/blob/develop/examples/mqtt-advanced/mqtt-advanced.ino

You can achieve your goal by adding a new helper, create automation for at and publish a custom MQTT message using MQTT service (https://www.home-assistant.io/docs/mqtt/service/). On the Arduino side, you just need to listen for your message (subscribe) and parse its payload.

closed 2021-04-17 by dawidchyrzynski