dawidchyrzynski/arduino-home-assistant #25: Multiple Sensors (NodeMCU)

gmarjano1975, 2021-04-01:
Hi,

I am trying to create a device with multiple sensors. I got a test sketch with is running.
MQTT.fx shows two "config"-Topics and periodically published two "state" topics:

- homeassistant/sensor/ecfabc594575/mq2/config
```
{
	"stat_t": "homeassistant/sensor/ecfabc594575/mq2/state",
	"dev_cla": "None",
	"unit_of_meas": "ppm",
	"ic": "mdi:gas-cylinder",
	"name": "mq2",
	"uniq_id": "mq2_ecfabc594575",
	"dev": {
		"ids": "ecfabc594575",
		"mf": "WeMos D1 Mini",
		"mdl": "NodeMCU",
		"name": "CO-Sensor",
		"sw": "1.0.0"
	}
}
``` 
- homeassistant/sensor/ecfabc594575/mq7/config
```
{
	"stat_t": "homeassistant/sensor/ecfabc594575/mq7/state",
	"dev_cla": "temperature",
	"unit_of_meas": "°C",
	"ic": "mdi:home",
	"name": "mq7",
	"uniq_id": "mq7_ecfabc594575",
	"dev": {
		"ids": "ecfabc594575",
		"mf": "WeMos D1 Mini",
		"mdl": "NodeMCU",
		"name": "CO-Sensor",
		"sw": "1.0.0"
	}
}
```

Every two seconds the state-topic is published:
- homeassistant/sensor/ecfabc594575/mq2/state
- homeassistant/sensor/ecfabc594575/mq2/state

Home Assistant auto-discovers the mq7 sensor. But not the second one (mq2). This is my sketch:
```
#include <ESP8266WiFi.h>
#include <ArduinoHA.h>

#define ARDUINOHA_DEBUG TRUE
#define WIFI_SSID  "xxx"
#define WIFI_PASSWORD  "xxx"
#define MQTT_BROKER  "xxx"
#define MQTT_USER  "xxx"
#define MQTT_PASS  "xxx"

unsigned long lastSentAt = millis();
double lastValue = 0;

WiFiClient client;
HADevice device;
HAMqtt mqtt(client, device);
HASensor mq7("mq7"); // you can use custom name in place of "led"
HASensor mq2("mq2"); 

void setup() {
    Serial.begin(9600);
    Serial.println("Starting...");

    // connect to wifi
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500); // waiting for the connection
    }
    Serial.println();
    Serial.println("Connected to the network");

    
    // Unique ID must be set!
    byte mac[WL_MAC_ADDR_LENGTH];
    WiFi.macAddress(mac);
    device.setUniqueId(mac, sizeof(mac));

    // set device's details (optional)
    device.setName("CO-Sensor");
    device.setSoftwareVersion("1.0.0");
    device.setModel("NodeMCU");
    device.setManufacturer("WeMos D1 Mini");

     // configure sensor (optional)
    mq7.setUnitOfMeasurement("°C");
    mq7.setDeviceClass("temperature");
    mq7.setIcon("mdi:home");

    mq2.setUnitOfMeasurement("ppm");    
    mq2.setDeviceClass("None");
    mq2.setIcon("mdi:gas-cylinder");

    
    mqtt.begin(MQTT_BROKER, MQTT_USER, MQTT_PASS);
}

void loop() {
    mqtt.loop();

    if ((millis() - lastSentAt) >= 5000) {
          Serial.println("Loop...");

        lastSentAt = millis();
        lastValue = lastValue + 0.5;
        mq2.setValue(lastValue);
        mq7.setValue(lastValue+1);
    }
    delay(2000);
}
```
I was trying to use `#define ARDUINOHA_DEBUG TRUE` but no debugging information is printed on serial. Am I missing something? 

Best regards
Goran

dawidchyrzynski, 2021-04-01:
Hi @gmarjano1975,

Most likely problem is caused by this line `mq2.setDeviceClass("None");`. Home Assistant requires "device class" to be one of the values listed here: https://www.home-assistant.io/integrations/sensor/

If you don't need to set class just remove this line. It's "none" by default.

gmarjano1975, 2021-04-04:
Hi Dawid,

thanks a lot. That helped. I removed `mq2.setDeviceClass("None");` and now it works.

Best regards
Goran

> Hi @gmarjano1975,
> 
> Most likely problem is caused by this line `mq2.setDeviceClass("None");`. Home Assistant requires "device class" to be one of the values listed here: https://www.home-assistant.io/integrations/sensor/
> 
> If you don't need to set class just remove this line. It's "none" by default.

gmarjano1975, 2021-04-05:
Hi Dawid,

maybe you got an idea why my ``onSwitchStateChanged`` is called after a restart.
```
void onSwitchStateChanged(bool state, HASwitch* s)
{

    if (state) {
      Serial.print("State: "); Serial.println("TRUE");
    } else {
      Serial.print("State: "); Serial.println("FALSE");
    }
    if (state) {      
      uint addr = 0;      
      int val = 0;
      EEPROM.begin(4);
      EEPROM.get(addr,val);
      if (val <= 0) {
        val =750;
        EEPROM.put(addr,val);
        EEPROM.commit();    
        Serial.print("To EEPORM: ");Serial.println(val);
      } else {
        Serial.print("From EEPORM: ");Serial.println(val);
      }
      
      

      calibrate();
      cal.turnOff();
    }
}
void setup() {
...
    // set device's details (optional)
    device.setName("CO-Sensor");
    device.setSoftwareVersion("1.0.0");
    device.setModel("NodeMCU");
    device.setManufacturer("WeMos D1 Mini");

     // configure sensor (optional)
    mq7.setUnitOfMeasurement("°C");
    mq7.setDeviceClass("temperature");
    mq7.setIcon("mdi:home");

    mq2.setUnitOfMeasurement("ppm");    
    mq2.setIcon("mdi:gas-cylinder");

    
    cal.onStateChanged(onSwitchStateChanged);   
    cal.setIcon("mdi:calculator");

    
    mqtt.begin(MQTT_BROKER, MQTT_USER, MQTT_PASS);
}
```

After a reset/reboot, the MQTT.fx shows one retained message where the switch is off. 
After some ms I got a message with the state = TRUE, but I don't know why and where it comes from.

I would only expect the first OFF-message. And when I hit the switch-Button in HA, the ON-message should arrive.
Maybe it's HA? 

[image]
15:05:54.972 -> Connected to the network
15:05:59.668 -> State: TRUE
15:05:59.668 -> From EEPORM: 750
15:05:59.668 -> Calibrating...done
15:06:04.654 -> State: FALSE
15:06:04.654 -> Loop...
15:06:10.650 -> Loop...
15:06:16.655 -> Loop...
15:06:22.660 -> Loop...
15:06:28.648 -> Loop...
15:06:34.663 -> Loop...
15:06:40.654 -> Loop...

BTW, do you have any ideas why the debug-flag is not working? :-/

Best regards
Goran

dawidchyrzynski, 2021-04-06:
If you want to use a switch with the retained flag you need to set retain mode for the switch by calling `cal.setRetain(true)`.

Regarding debug flag. How do you set it? Try my instructions described here: https://github.com/dawidchyrzynski/arduino-home-assistant/issues/20#issuecomment-804060325

gmarjano1975, 2021-04-07:
Hi Dawid,

actually I did not want a retained message for the switch. I guess I had it configured retained in the first place. And my test was to click on the switch in home assistant -> First message and Mosquitto stored it.
I deleted the retained topic in Mosquitto and configured the switch without retain. Now it works. After a restart of my wemos 
there is no "ON" message.

Debugging worked after I added the define-line in the correct file. I had it in my sketch, so it did not work.

Thanks again.

Best regards
Goran

closed 2021-04-07 by gmarjano1975