dawidchyrzynski/arduino-home-assistant #26: Suscribing to a custom topic

netixx, 2021-04-01:
Hello,

For my use case (which looks like [this](https://esphome.io/components/deep_sleep.html)), I want to subscribe to a custom topic and react accordingly.

Since we already provide the necessary parameters to HAmqtt, I wanted to use this object to subscribe to the topic with a custom function to execute upon reception of the message.

What would be the recommended way to do this with this library ? Should I implement a custom `DeviceType` ?

dawidchyrzynski, 2021-04-01:
Hi @netixx,

You can subscribe to any topic using `bool subscribe(const char* topic)` method from HAMqtt class, but I just realized it's not possible to register a callback for receiving messages. I will fix this soon.

dawidchyrzynski, 2021-04-12:
Hi @netixx

This feature is available on the develop branch. You can check example here: https://github.com/dawidchyrzynski/arduino-home-assistant/blob/develop/examples/mqtt-events/mqtt-events.ino

closed 2021-04-17 by dawidchyrzynski