dawidchyrzynski/arduino-home-assistant #44: Add device class and icons for covers

brainrecall, 2021-07-07:
Hi, awesome library that will save me a bunch of time. I'm creating a garage door controller and it would be awesome if you were to add the `setDeviceClass()` and `setIcon()` methods to `HACover` like you have for the `HASensor`. Is that possible?

incaseoftrouble, 2021-08-04:
I'd propose the following changes to HACover:
 * features to indicate support of position and/or tilt
 * callbacks for position / tilt set by HA
 * `setTilt`, `setPositionAndTilt` methods

Since `icon` and `device_class` is something common to all entities (class is only supported by some, but the field exists on base entities), it might make sense to simply pull these methods to the base device type.

@dawidchyrzynski I can try to implement these things if you agree (similar things exist in other classes, so its mostly copy&paste)

dawidchyrzynski, 2022-06-04:
This feature/fix will be implemented as part of the new major release - see #81

closed 2022-06-04 by dawidchyrzynski