dawidchyrzynski/arduino-home-assistant #46: dht sensors problem

tremme, 2021-08-03:
hello I would like to put 7 sensors dht then 14 sensors in has between temperature and humidity
but if I exceed 8 sensors it does not recognize the device in the home assistant
how can it be solved ??
is there a limit on the number of sensors ??

dawidchyrzynski, 2021-08-28:
Hi @tremme,

The only limitation is amount of available RAM. If you're using Arduino UNO for your project then you should really consider upgrading it to Arduino Mega or ESP boards. Can you provide more details about your project?

zemaservice, 2021-10-11:
hello I noticed that the problem is on mosquitto 
and ' as if you keep in memory that arduino and' connected 
but instead it is not
so arduino tries to connect but fails
let me know if there is a control method
thank you

pvoncken, 2021-11-10:
> hello I would like to put 7 sensors dht then 14 sensors in has between temperature and humidity but if I exceed 8 sensors it does not recognize the device in the home assistant how can it be solved ?? is there a limit on the number of sensors ??

Hello,
I working on a project with 8 relays (or switches) 2 fans and 8 sensors.
I have the relays and 1 fan working but i have troubles with the second fan and the sensors.
Wouls you mind to share your sketch? Maybe it helps me.

tremme, 2021-11-10:
#include <Ethernet.h>

#include <ArduinoHA.h>

#include <SimpleDHT.h>

 

 

//// argomento da ascoltare   homeassistant/sensor/#

 

//#define LED_PIN             9

#define BROKER_ADDR         IPAddress(192,168,1,173)

#define BROKER_USERNAME     "*****" // replace with your credentials

#define BROKER_PASSWORD     "******”

 

unsigned long lastSentAt_dht = millis();

unsigned long lastSentAt_mqtt = millis();

 

byte mac[] = {0x00, 0x10, 0xFA, 0x6E, 0x38, 0x4A};

 

#define DHTPIN1 2  // definisco pin dht

#define DHTPIN2 3

#define DHTPIN3 0

#define DHTPIN4 5

#define DHTPIN5 6

#define DHTPIN6 7

#define DHTPIN7 8

#define DHTPIN8 1

 

SimpleDHT22 dht0(DHTPIN1);

SimpleDHT22 dht1(DHTPIN2);

SimpleDHT22 dht2(DHTPIN3);

SimpleDHT22 dht3(DHTPIN4);

 

SimpleDHT22 dht4(DHTPIN5);

SimpleDHT22 dht5(DHTPIN6);

SimpleDHT22 dht6(DHTPIN7);

SimpleDHT22 dht7(DHTPIN8);

[... 492 lines cut ...]
     if (prog == 7 )  {

        

        temperature[7] = 0 ;

        humidity[7] = 0 ;

        int err = SimpleDHTErrSuccess;

        if ((err = dht7.read2(&temperature[7], &humidity[7], NULL)) != SimpleDHTErrSuccess) {delay(2500);}

//        dht7.read2(&temperature[7], &humidity[7], NULL); 

        temp7.setValue((float)temperature[7]);

        umi7.setValue((float)humidity[7]);

       

//        prog = 0;

}                                 

}

pvoncken, 2021-11-10:
> hello I would like to put 7 sensors dht then 14 sensors in has between temperature and humidity but if I exceed 8 sensors it does not recognize the device in the home assistant how can it be solved ?? is there a limit on the number of sensors ??

Many Thanxs!!

closed 2022-06-04 by dawidchyrzynski