dawidchyrzynski/arduino-home-assistant #48: know which HASwitch that triggered a callback ??

tv4you2016, 2021-08-17:
Hello.
I have the following question:
I have 2 switches pointing to the same calback

 led.onBeforeStateChanged(onBeforeSwitchStateChanged); // optional
  led.onStateChanged(onSwitchStateChanged);

  led2.onBeforeStateChanged(onBeforeSwitchStateChanged); // optional
  led2.onStateChanged(onSwitchStateChanged);
  
is there any way to callback
void Before Switch StateChanged (bool state, HASwitch * s)
and
void onSwitchStateChanged (bool state, HASwitch * s)

know which HASwitch that triggered a callback ??

dawidchyrzynski, 2021-08-28:
Hi @tv4you2016,

The second argument passed to the callback is instance of the HASwitch that called it. You can identify switches as following:
```cpp
void onSwitchStateChanged(bool state, HASwitch* s) {
     // "s" is instance of the switch that called the callback
     Serial.println(s->uniqueId());
}
```

dawidchyrzynski, 2021-08-28:
You can also compare address of the pointer with instance of your switch. Example:

```cpp
HASwitch ledA("ledA", false);
HASwitch ledB("ledB", false);

void onSwitchStateChanged(bool state, HASwitch* s)
{
    if (s == &ledA) {
        Serial.println("Called by A");
    } else if (s == &ledB) {
        Serial.println("Called by B");
    }
}

void setup() {
    ledA.onStateChanged(onSwitchStateChanged);
    ledB.onStateChanged(onSwitchStateChanged);
}

```

closed 2021-08-28 by dawidchyrzynski

pvoncken, 2021-10-30:
> You can also compare address of the pointer with instance of your switch. Example:
> 
> ```c++
> HASwitch ledA("ledA", false);
> HASwitch ledB("ledB", false);
> 
> void onSwitchStateChanged(bool state, HASwitch* s)
> {
>     if (s == &ledA) {
>         Serial.println("Called by A");
>     } else if (s == &ledB) {
>         Serial.println("Called by B");
>     }
> }
> 
> void setup() {
>     ledA.onStateChanged(onSwitchStateChanged);
>     ledB.onStateChanged(onSwitchStateChanged);
> }
> ``

Hello,

I succesfullyt used this method on 12 Relays (or Switches).
Now I wouls like to do the same with 4 Fans.
I constructed the 4 Fans and they are available in Home Assistant. 
But i don't understand how I can Write the differnt fans to differnt PIN's.
I managed to write the State and Speed of one Fan to a Digital and an Analog Pin.
But i don't understand how ik can do the same for the other Fans.
I thougt that the method above could be used but at the moment I'm stuck.