dawidchyrzynski/arduino-home-assistant #75: HAMqtt.h error - Client does not name a type

Friend-Or-F0H, 2022-02-25:
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "Arduino NANO 33 IoT"

Header lines used:

#include <ArduinoBLE.h>
#include <WiFiNINA.h>
#include <ArduinoHA.h>
#include <OneWire.h>

#define LED_PIN                  13
#define BROKER_ADDR       IPAddress(192,168,0,17)
#define WIFI_SSID               "$myNetworkSSID"
#define WIFI_PASSWORD   "$myNetworkPASS"

WiFiClient client;
HADevice device;
HAMqtt mqtt(client, device);

Errors below:

In file included from M:\Basso\Documents\Arduino\libraries\home-assistant-integration\src\device-types\HAFan.cpp:4:0:

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token

     HAMqtt(Client& netClient, HADevice& device);

                  ^

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?

     Client& _netClient;

     ^~~~~~

     DNSClient

In file included from M:\Basso\Documents\Arduino\libraries\home-assistant-integration\src\device-types\HASensor.cpp:8:0:

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token

     HAMqtt(Client& netClient, HADevice& device);

                  ^

In file included from M:\Basso\Documents\Arduino\libraries\home-assistant-integration\src\device-types\HACover.cpp:4:0:

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token

     HAMqtt(Client& netClient, HADevice& device);

                  ^

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?

     Client& _netClient;

     ^~~~~~

     DNSClient

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?

     Client& _netClient;

     ^~~~~~

     DNSClient

In file included from M:\Basso\Documents\Arduino\libraries\home-assistant-integration\src\device-types\HABinarySensor.cpp:5:0:

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token

     HAMqtt(Client& netClient, HADevice& device);

                  ^

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?

     Client& _netClient;

     ^~~~~~

     DNSClient

In file included from M:\Basso\Documents\Arduino\libraries\home-assistant-integration\src\device-types\BaseDeviceType.cpp:2:0:

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token

     HAMqtt(Client& netClient, HADevice& device);

                  ^

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?

     Client& _netClient;

     ^~~~~~

     DNSClient

In file included from M:\Basso\Documents\Arduino\libraries\home-assistant-integration\src\device-types\HATriggers.cpp:5:0:

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token

     HAMqtt(Client& netClient, HADevice& device);

                  ^

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?

     Client& _netClient;

     ^~~~~~

     DNSClient

In file included from M:\Basso\Documents\Arduino\libraries\home-assistant-integration\src\device-types\HATagScanner.cpp:5:0:

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token

     HAMqtt(Client& netClient, HADevice& device);

                  ^

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?

     Client& _netClient;

     ^~~~~~

     DNSClient

In file included from M:\Basso\Documents\Arduino\libraries\home-assistant-integration\src\device-types\HASwitch.cpp:5:0:

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token

     HAMqtt(Client& netClient, HADevice& device);

                  ^

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?

     Client& _netClient;

     ^~~~~~

     DNSClient

In file included from M:\Basso\Documents\Arduino\libraries\home-assistant-integration\src\device-types\HAHVAC.cpp:5:0:

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token

     HAMqtt(Client& netClient, HADevice& device);

                  ^

m:\basso\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?

     Client& _netClient;

     ^~~~~~

     DNSClient

exit status 1

Error compiling for board Arduino NANO 33 IoT.

Omnipius, 2022-04-05:
I am experiencing an identical error using an Arduino Uno WiFi Rev2.

sushukka, 2022-04-10:
With Nano 33 IoT downgrade your Arduino SAM board libraries to 18. Same with WifiNINA software libraries -> 18.1.

dawidchyrzynski, 2022-06-04:
This feature/fix will be implemented as part of the new major release - see #81

closed 2022-06-04 by dawidchyrzynski