dawidchyrzynski/arduino-home-assistant #77: Code blocks when home assistant is disconnected

mrunmaydarpan, 2022-03-07:


dawidchyrzynski, 2022-06-04:
Hi @mrunmaydarpan,

Can you provide more details? Which board do you use?

mrunmaydarpan, 2022-06-05:
> 

I was using esp12e

milansim, 2023-01-18:
I have the same problem when wifi disconnected or home assistant not present...on NodeMCU32s and Wemos D1mini
I built it to control lights and roller shutters so it MUST work in " offline mode"
mqtt.loop(); is blocking in loop becouse it takes approx 7 seconds when mqtt.loop tries to reconnect
My temporary solution is

void mqtt_240s_loopcheck() {
    if (currentMillis - mqttmillis >= 240000) { mqttmillis = currentMillis; telnet.println("Mqtt 240s loopcheck"); mqtt.loop();} // svako 240 sekundi da ne blokira loop ako neradi homeassistant...traje 7 sec pokušaj spajanja
 }
void loop() {...
  wifi_chk(); // my wifi check every 5 secs
  if (wifiConnected == true && mqtt.isConnected() == true) { mqtt.loop(); mqttmillis = currentMillis; HA_connected = true; } 
  else { HA_connected = false; if (wifiConnected == true) {mqtt_240s_loopcheck()} }

Doing it this way HA has a chance to connect every 4 minutes and block code 7 secs, otherwise code is not blocking
Real solution lies in pubsubclient non blocking example but I dont know how to implement with this library

DzejbiB, 2023-03-27:
Hi,

in my setup:
library 'home-assistant-integration' v. 2.0.0
library 'PubSubClient' v. 2.8.0
board: Arduino Mega

when Home Assistant is disconnected mqtt.loop() is blocked for 1000 milliseconds (most probably it's a connection timeout) every 5 seconds (because ReconnectInterval=5000).

 I have light and cover switches on the wall and additonally I'm using Home Assistant. If HA is not avaliable, switches on the wall don't work as expected because mqtt.loop(); is blocked for 1s too frequent. Would it be possible to add API to change ReconnectInterval value (to e.g. 5 or 10 minutes)?

erkr, 2023-11-27:
I call mqtt.loop() only if there is a MQTT connection.