dawidchyrzynski/arduino-home-assistant #82: Not compiling

bogdan2000, 2022-06-14:
On Arduino IDE 2.0.0.-rc3 when compiling on Nano 33 IOT your examples I get:

" In file included from C:\Users\Lenovo Yoga\Documents\Arduino\libraries\home-assistant-integration\src\device-types\HAFan.cpp:4:0:
c:\users\lenovo yoga\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token
     HAMqtt(Client& netClient, HADevice& device);
                  ^
c:\users\lenovo yoga\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?
     Client& _netClient;
     ^~~~~~
     DNSClient
In file included from C:\Users\Lenovo Yoga\Documents\Arduino\libraries\home-assistant-integration\src\device-types\HABinarySensor.cpp:5:0:
c:\users\lenovo yoga\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token
     HAMqtt(Client& netClient, HADevice& device);
                  ^
c:\users\lenovo yoga\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?
     Client& _netClient;
     ^~~~~~
     DNSClient
In file included from C:\Users\Lenovo Yoga\Documents\Arduino\libraries\home-assistant-integration\src\device-types\HACover.cpp:4:0:
c:\users\lenovo yoga\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token
     HAMqtt(Client& netClient, HADevice& device);
                  ^
c:\users\lenovo yoga\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?
     Client& _netClient;
     ^~~~~~
     DNSClient
In file included from C:\Users\Lenovo Yoga\Documents\Arduino\libraries\home-assistant-integration\src\device-types\BaseDeviceType.cpp:2:0:
c:\users\lenovo yoga\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token
     HAMqtt(Client& netClient, HADevice& device);
                  ^
c:\users\lenovo yoga\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?
     Client& _netClient;
     ^~~~~~
     DNSClient
In file included from C:\Users\Lenovo Yoga\Documents\Arduino\libraries\home-assistant-integration\src\device-types\HAHVAC.cpp:5:0:
c:\users\lenovo yoga\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:23:18: error: expected ')' before '&' token
     HAMqtt(Client& netClient, HADevice& device);
                  ^
c:\users\lenovo yoga\documents\arduino\libraries\home-assistant-integration\src\hamqtt.h:224:5: error: 'Client' does not name a type; did you mean 'DNSClient'?
     Client& _netClient;
     ^~~~~~
     DNSClient

Compilation error: exit status 1 "

dawidchyrzynski, 2022-06-14:
Hi @bogdan2000,

Please provide source code and information about platform you use (Arduino Uno / ESP / etc.).

bogdan2000, 2022-06-14:
It's your example for Nano 33 IoT but it's verified under Arduino IDE 2.0.0.-rc3

#include <WiFiNINA.h>
#include <ArduinoHA.h>

#define LED_PIN         9
#define BROKER_ADDR     IPAddress(192,168,0,17)
#define WIFI_SSID       "MyNetwork"
#define WIFI_PASSWORD   "MyPassword"

WiFiClient client;
HADevice device;
HAMqtt mqtt(client, device);
HASwitch led("led", false); // "led" is unique ID of the switch. You should define your own ID.

void onSwitchStateChanged(bool state, HASwitch* s)
{
    digitalWrite(LED_PIN, (state ? HIGH : LOW));
}

void setup() {
    Serial.begin(9600);
    Serial.println("Starting...");

    // Unique ID must be set!
    byte mac[WL_MAC_ADDR_LENGTH];
    WiFi.macAddress(mac);
    device.setUniqueId(mac, sizeof(mac));

    pinMode(LED_PIN, OUTPUT);
    digitalWrite(LED_PIN, LOW);

    // connect to wifi
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500); // waiting for the connection
    }
    Serial.println();
    Serial.println("Connected to the network");

    // set device's details (optional)
    device.setName("Nano 33 IoT");
    device.setSoftwareVersion("1.0.0");

    // handle switch state
    led.onStateChanged(onSwitchStateChanged);
    led.setName("My LED"); // optional

    mqtt.begin(BROKER_ADDR);
}

void loop() {
    mqtt.loop();
}

shall0pass, 2022-07-16:
I get the same errors trying to compile the example.

bogdan2000, 2022-07-17:
In src/HAMqtt.h file I modified 

 HAMqtt(Client& netClient, HADevice& device); 
to
HAMqtt(arduino::Client& netClient, HADevice& device);
and 
Client& _netClient;
to 
arduino::Client& _netClient;

Tell me if it works.

stevetemple, 2022-08-08:
That change works for me also, clashing namespace?

dawidchyrzynski, 2022-09-05:
This issue will be fixed as part of the #81

closed 2022-09-05 by dawidchyrzynski