dawidchyrzynski/arduino-home-assistant #86: Arduino cycle restart - limit of values?

proasnet, 2022-06-27:
Hello, I made some projects with this library. All working fine, but the last project has more values, and library restarting cpu in cycle.
I try comment all library codes and other working.
Now, in project is 25 analog sensors and 7 binary sensors.
Have you any idea, where may be a problem?
Thank you very much

dawidchyrzynski, 2022-07-03:
Hi @proasnet,

It sounds like you're hitting the resources limit (RAM). Which platform do you use?

proasnet, 2022-07-03:
Hello. I found problem. Some values was declared as byte or unsigned long
int. I found, that your library does not support this types, only int,
float or bool.
This restarts the arduino.
Long int I must declare as float and byte as int and work fine.
Thank you.

closed 2022-07-16 by dawidchyrzynski