dawidchyrzynski/arduino-home-assistant #89: Request: Multi-mode switch

daniloc, 2022-08-09:
Hello! Thanks for a great project.

I am integrating a Mitsubishi Heat Pump into Home Assistant, via IR signaling, and this project as made for a great start.

I've had success by composing multiple device types so far (`HAHVAC` and `HASensor`). But one primitive does not seem to exist in this library:

A multi-mode switch.

In the case of the Heat Pump, this would support things like:

- Fan Mode (1-5, auto, silent)
- Horizontal and vertical vane mode (series of angles, sweep, breeze)

Can you recommend a good direction for toggling between a series of integers, which I'll interpret as enum values for the various modes?

daniloc, 2022-08-09:
Digging deeper, it seems that there's built in support in both Home Assistant's [Climate integration](https://www.home-assistant.io/integrations/climate/#service-climateset_fan_mode), and in MQTT's [Climate implementation](https://www.home-assistant.io/integrations/climate.mqtt/), for both fan mode and swing mode.

So perhaps these enhancements would better target this library's HVAC implementation instead? I am finding the serialization mechanics a bit hard to follow, but it seems that @dawidchyrzynski is hard at work on enhancements on exactly that sub-system.

As I see you have not yet committed anything related to HVAC in that branch, @dawidchyrzynski, might I request you consider these additional functions when you get a moment for the HVAC device implementation?

Thanks so much for your time on all of this, it's such a great project.

dawidchyrzynski, 2022-08-20:
Hi @daniloc,

I will research this feature and eventually implement it as part of the coming release - https://github.com/dawidchyrzynski/arduino-home-assistant/pull/81

dawidchyrzynski, 2022-10-07:
Hi @daniloc,

The new HAHVAC implementation is already available on the `serializer-refactor` branch. Feel free to give it a go.

closed 2022-10-14 by dawidchyrzynski