dawidchyrzynski/arduino-home-assistant #99: MQTT.isConnected() always return false

MingyaoLiu, 2022-09-29:
Hi,
I'm just wondering if this is a singular case for me, but I check mqtt.isConnected in loop before calling mqtt.loop(), and isConnected always returns false.

I'm on arduino-pico with a pico w. If I don't check isConnected, mqtt works fine, but if there is any issue with my internet, mqtt.loop will crash the pico.

MingyaoLiu, 2022-09-29:
I have figured out a workaround.

What happened is if MQTT.loop is not called at least once, MQTT.isConnected will always fail. Not sure if this is a bug or desired behaviour, but workaround is just to call mqtt.loop once in setup.

MingyaoLiu, 2022-09-29:
A simple workaround example:

```
void setup() {
    if (!mqtt.begin(MQTT_ADDR, MQTT_PORT, MQTT_LOGIN, MQTT_PASSWORD)) {
    Serial.print("Could not connect to MQTT broker.");
  } else {
    Serial.println("Connected to MQTT broker.");
    mqtt.loop();
  }
}
void loop() {
    while (!mqtt.isConnected()) {
    Serial.println("MQTT not connected in loop, try re-connecting...");
    mqtt.disconnect(false);
    if (!mqtt.begin(MQTT_ADDR, MQTT_PORT, MQTT_LOGIN, MQTT_PASSWORD)) {
      Serial.println("Could not re-initiate MQTT service.");
    } else {
      mqtt.loop();
    }
    delay(1000);
  }
  mqtt.loop();
}
```

dawidchyrzynski, 2022-09-30:
Hi @MingyaoLiu,

That's correct behaviour. The `mqtt.begin(...)` schedules asynchronous connection. The connection attempt is performed during the first loop cycle.

closed 2022-09-30 by dawidchyrzynski