dawidchyrzynski/arduino-home-assistant #101: Request: RemoteReceiver

corvin78, 2022-10-01:
Hi,

At first, thank You for great project!

I am trying to use infrared and rf (315MHz) remote controllers with my homemade relay controller. Thanks to your library I can flip relays from HA as well. I did implement remote controll of my relays with Arduino-IRremote library and it works perfectly fine for flipping relays, but I would really like to get the remote controll presses back to HA for automations with other devices. Is there a possibility to add that funcionality to your library?

Regards,
Łukasz

dawidchyrzynski, 2022-10-04:
Hi @corvin78,

The main and only purpose of this library is to provide an easy-to-use wrapper for MQTT communication. I can implement an example with an IR controller as part of the coming release (#81).

corvin78, 2022-10-04:
Hi,

Thank you, simple example would be great help!

Regards,

romoloman, 2024-05-19:
@corvin78  
Basically you have to define a switch device and set it's state depending on the remote control code received.
have a look to the setstate() method switch object.