dawidchyrzynski/arduino-home-assistant #103: Dynamic entries creation?

Elektrik1, 2022-10-03:
Hey there,
I am interested whether it is possible to create entries (particularly HASwitch) dynamically - in loop().
Tried to initiate HASwitch class in loop - no errors appear, but no data to MQTT and HA is sent.
Is there any option to add entries dynamically on the fly?

dawidchyrzynski, 2022-10-04:
Hi @Elektrik1,

Can you provide more details about your use case? Theoretically, it should be possible if the entity remains in the memory for a lifetime.

Elektrik1, 2022-10-05:
Assume we have such program:
```
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <PubSubClient.h>
#include <ArduinoHA.h>

WiFiClient espClient;

HADevice device;
HAMqtt mqtt(espClient, device);

bool created = false;

void setup() {
  byte mac[WL_MAC_ADDR_LENGTH];
  WiFi.macAddress(mac);
  device.setUniqueId(mac, sizeof(mac));

  device.setName("Test");
  device.setSoftwareVersion("1.0.0");
  device.setModel("TP01");

  mqtt.begin("192.168.1.1");
}

void loop() {
  mqtt.loop();

  if(!created){
    HASwitch led("led", false);
    led.setName("My LED");
    created = true;
  }
}
```

It works fine if I declare HASwitch in setup(), before mqtt.begin, however if I create HASwitch instance within the loop() dynamically - it does not appear in HA and no errors are produced. The problem is HASwitch count changes during the loops (it is configurable externally), therefore I ask if there is a possibility to create HASwitch instances on the fly.

Elektrik1, 2022-10-05:
It works! Thank you very much, that is exactly what I tried to do.

Elektrik1, 2022-10-26:
I am reopening this, because I think it is related. 
There is a problem, that if I create HADevice at runtime dynamically (after mqtt is connected), there is no way to force config publish (`publishConfig()` in base in protected), therefore I found no way to signal the HA about new device, and the callback is not received by `onCommand()`. I think it should be exposed the same way as `setAvailability()` is.

rikki78, 2022-11-19:
Hi,
When dynamically creating these sensors and naming it, the name given needs to remain available for the library. So when using the same buffer (for example to create more sensors at a time and naming them), all sensors will get the same name. Is there a way to make the library copy the name so the application can make dynamic names?

EmilEmilchen, 2023-01-14:
> 

For some reason basically the same code doesnt work for me. It always gives `error: no matching function for call to 'HASwitch::HASwitch(const char [4], bool)'`. Im using the following Code. `#include <Ethernet.h>
#include <ArduinoHA.h>

#define LED_PIN         9
#define BROKER_ADDR     IPAddress(192,168,0,17)

byte mac[] = {0x00, 0x10, 0xFA, 0x6E, 0x38, 0x4A};

EthernetClient client;
HADevice device(mac, sizeof(mac));
HAMqtt mqtt(client, device);
HASwitch* switchPtr = nullptr;

// "led" is unique ID of the switch. You should define your own ID.

void onSwitchCommand(bool state, HASwitch* sender)
{
    digitalWrite(LED_PIN, (state ? HIGH : LOW));
    sender->setState(state); // report state back to the Home Assistant
}

void setup() {
    pinMode(LED_PIN, OUTPUT);
    digitalWrite(LED_PIN, LOW);

    // you don't need to verify return status
    Ethernet.begin(mac);

    // set device's details (optional)
    device.setName("Arduino");
    device.setSoftwareVersion("1.0.0");

    mqtt.begin(BROKER_ADDR);
}

void loop() {
    Ethernet.maintain();
    mqtt.loop();

    if(!switchPtr){
      switchPtr = new HASwitch("led", false);
      switchPtr->setName("My LED");
    }
}
`