dawidchyrzynski/arduino-home-assistant #105: 2.0.0 discovery issue

Enlumis, 2022-10-20:
Hello,
since i updated my code to 2.0.0 i got discovery issue, 
even using MQTT Explorer i cannot see the complete list of topics
i only see the first topic here is some code
`
HASwitch ledSWITCH_1_1("NSWITCH_1_1"); 
HASwitch ledSWITCH_1_2("NSWITCH_1_2");
HASwitch ledSWITCH_1_3("NSWITCH_1_3"); 
HASwitch ledSWITCH_1_4("NSWITCH_1_4"); 
HASwitch ledSWITCH_1_5("NSWITCH_1_5"); 
....
setup() {
  ledSWITCH_1_1.setIcon("mdi:lightbulb");
    ledSWITCH_1_1.setName("light_1_1");
    ledSWITCH_1_1.onCommand(onSwitchStateChanged);

    ledSWITCH_1_2.setIcon("mdi:lightbulb");
    ledSWITCH_1_2.setName("light_1_2");
    ledSWITCH_1_2.onCommand(onSwitchStateChanged);

    ledSWITCH_1_3.setIcon("mdi:lightbulb");
    ledSWITCH_1_3.setName("light_1_3");
    ledSWITCH_1_3.onCommand(onSwitchStateChanged);
    ...
    }
    
    
void onSwitchStateChanged(bool state, HASwitch* s)
{
  char* uid = s->uniqueId();
  for(uint8_t i = 0; i < sizeof(theSwitchMap)/sizeof(switchMap); ++i) {
      if (strcmp(theSwitchMap[i].uid, uid) == 0) {
        digitalWrite(theSwitchMap[i].pin,  HIGH );
        delay(200);
        digitalWrite(theSwitchMap[i].pin,  LOW);
        s->setState(state);
      }
  }
}
`
basically the issue is that i only see one discovered topic "NSWITCH_1_1"
thats it. other are ignored gone

i also got an issue with HACover now, i cannot setPosition even using your new second parameter flag to force it
thats unfortunate because i got rts cover so i dont get feedback from it, i would like to have the UP/DOWN button to be always clickable in homeassistant so i was forcing the position to 50 no matter what was clicked
now the buttons in home assistant are not both clickable, not sure how to get around about that one. 
Im more worry about the discovery issue, i might just revert back to 1.3.0
`
void onCoverCommand1(HACover::CoverCommand cmd, HACover* sender) {
    if (cmd == HACover::CommandOpen) {
        Serial.println("Command: Open");
        sender->setState(HACover::StateOpen);
        upVollet(0);
    } else if (cmd == HACover::CommandClose) {
        Serial.println("Command: Close");
        sender->setState(HACover::StateClosed);
        downVollet(0);
    } else if (cmd == HACover::CommandStop) {
        Serial.println("Command: Stop");
        sender->setState(HACover::StateStopped);
        stopVollet(0);
    }
    sender->setPosition(50, true);

}`

dawidchyrzynski, 2022-10-21:
Hi @Enlumis,

How many switches are you trying to register in your real code? I tested the lib with 5 switches and it works fine.

HACover was refactored and the position feature is disabled by default. You need to enable it using the second argument of the HACover's constructor as follows: `HACover cover("myCover", HACover::PositionFeature);`

Enlumis, 2022-10-21:
Hi @dawidchyrzynski 
i got 18switches registered, i confirm rolling back to 1.3.0 fixed it
thanks for the cover ill try that
i also find out a HA fix for RTS covers
_configuration.yaml_
```homeassistant:
  customize_glob:
    "cover.*":
      assumed_state: true
```

dawidchyrzynski, 2022-10-21:
@Enlumis Please take a look at the documentation: https://dawidchyrzynski.github.io/arduino-home-assistant/documents/library/device-types.html

There is a default limit of 6 device types that can be registered at once. Please set it to 20 and check if it works. One more question. What platform do you use (Arduino Mega, ESP, etc.)?

Enlumis, 2022-10-21:
Thanks ill try that later,
I use Arduino Mega 2560

dawidchyrzynski, 2022-10-30:
@Enlumis any update?

closed 2022-11-04 by dawidchyrzynski

rmeissn, 2023-01-07:
I got 21 sensors in one device (on esp8266) and that works. Discovery on HA just took some time (several minutes) - I also wondered whether I'm affected by this "bug"... But it seems to work.