dawidchyrzynski/arduino-home-assistant #106: It is possible to create a multi device system?

antoniomolram, 2022-10-29:
Hi! First of all, thank you for your amazing library! 

I have your library working with multiple sensors (HABinarySensor, HASensorNumber) and I would like to use it as a mean to create on HA multiple devices. (I am creating a kind of a wireless hub). After checking your code, means a relative big rework, isn't it? 

Thank you for your help and comments. 
Regards,
Antonio

dawidchyrzynski, 2022-10-30:
Hi @antoniomolram,

There is no way to configure multiple device entities on the same physical device. Can you provide more details about your use case?

antoniomolram, 2022-10-30:
I have several lora devices reporting data, and I was thinking to have kind of a "gateway" converting all these lora devices HA in devices using your library. I have for the moment one, but I would love to have multiple HA devices created out of a single device. 

Do you think that this make sense?

rikki78, 2022-11-18:
Have a look at Mysensors, that is exactly doing this.

antoniomolram, 2022-11-18:
thank you! I will take a look to it! ;)

sozonnyk, 2023-04-14:
@dawidchyrzynski 
I have another use case: I have a single ESP32 which performs two logically different functions. So I'd like to have them separated in the HA interface as well. 

@antoniomolram  
I've used MySensors for a while, and it may be enough for your purposes. But in my case I want to have more logic on the physical, battery powered device. Thus I wanted to circumvent MySensors limitations by doing exactly what you proposed and controlling my HA devices from my own gateway. Its a bummer that multi devices are not supported.

chinswain, 2024-05-19:
My use case for this is dealing with multiple beehives.
I have 10 in an Apiary that transmit their data to a master hive, which then uses LoRa to transmit the sensor data for each hive (Prefixed with that hives ID number).

Ideally I'd then have a device for each beehive and update the sensor data for only that one, instead the only way is to have every hive's sensor as a unique sensor in HA: Hive2Temperature, Hive2Humidity etc.