dawidchyrzynski/arduino-home-assistant #112: Multiple sensors @ESP32

ssditt, 2022-11-20:
All my recent projects work fine with UNO or ESP8266. Thanks Dawid!
Today, I need 6 sensors @ESP32:
----------------------------------------------------------------
#include <ArduinoHA.h>
#include <WiFi.h>

#define BROKER_ADDR   "192.168.1.10"
#define MAC_ADDR_LENGTH 6

byte mac[] = "NAMFV1";
const char *unitName = "NAMFV1";
const char *unitVersion = "A1";
char WiFi_SSID [] = "ssid";
char WiFi_pasw [] = "pswd";
WiFiClient client;
HADevice device(mac, MAC_ADDR_LENGTH);
HAMqtt mqtt(client, device);

HASensor data("V");
HASensor dataL("A");
HASensor wattage("W");
HASensor voltageL("VL");
HASensor amperageL("AL");
HASensor wattageL("WL");

void setup()
{
  Serial.begin(9600);
  Serial.println("Starting...");
  WiFi.begin(WiFi_SSID, WiFi_pasw);
  while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500); 
  }
  Serial.println(" Connected to the network");

  device.setName(unitName);
  device.setSoftwareVersion(unitVersion);
  
  mqtt.begin(BROKER_ADDR);
}

void loop()  {
  mqtt.loop();
} 
---------------- ---------------------------------------

But I get only 5 sensors to MQTT - the last one is missing. And all stat_t contexts are defined with "aha" prefix:
{
  "uniq_id": "V",
  "dev": {
    "ids": "4e414d465631",
    "name": "NAMFV1",
    "sw": "A1"
  },
  "stat_t": "aha/4e414d465631/V/stat_t"
}

Finally I can get it worked merging sensor data into less sensors and it works with different context for stat_t, but it should be great to have standard functionality @ESP32

rikki78, 2022-11-20:
https://dawidchyrzynski.github.io/arduino-home-assistant/documents/library/device-types.html
By default the number of devices is 6. Did you try setting it?

ssditt, 2022-11-20:
I have got all 6 sensors in MQTT with the line
    HAMqtt mqtt(client, device,8);

Any idea what's wrong with prefix "aha" in stat_t context? It is still there with all sensors...

ssditt, 2022-11-20:
rikki78, thanks for the link! 
I have found the "aha" is a default prefix for data (stat_t) context. It is a new against older versions.

So the conclusion is that despite defualt limit 6 devices the program registers 5 devices only.

closed 2022-11-20 by ssditt

dawidchyrzynski, 2022-11-20:
Hi @ssditt,

The `aha` prefix is a configurable data prefix. It's described here: https://dawidchyrzynski.github.io/arduino-home-assistant/documents/library/discovery.html

If it comes to limit, there is a known bug in the entities limiter that will be fixed soon. For now please use a limit higher by one (in your case it should be `7`).