dawidchyrzynski/arduino-home-assistant #124: setUniqueId function for device types

simie7, 2023-01-08:
Hey, thanks for awesome library, love from first sight!

I am making few sensors at home on Wemos D1, that share the same setup, like the same pins, the same set of sensors (let's say DHT22 and DS18B20). I want to have same code for all, and it looks like there is no problem, as ESP has unique MAC or CHIP ID.

```
HADevice device;

void setup() {
// ...
  WiFi.macAddress(mac);
  device.setUniqueId(mac, sizeof(mac));

  sprintf(hostname, "wSensor-%08X", ESP.getChipId());
  device.setName(hostname);
// ...
}
```
I want to have the same possibility for device types, but I have to declare device unique ids before setup:
```
HASensorNumber bs18d20("dallas-temperature", HASensorNumber::PrecisionP1), 
               temp("dht-temperature", HASensorNumber::PrecisionP1), 
               hum("dht-humidity", HASensorNumber::PrecisionP1);

```
It is possible get chip id in variables declaration section, but then it's complex to do something with uint_32 as prefix and nice human readable name like `temperature`. 

I've checked how `Tasmota` handles it, here is example:
```
homeassistant/sensor/B3F48A_AM2301_Temperature/config {"name":"wTasmota01 AM2301 Temperature","stat_t":"tele/tasmota_B3F48A/SENSOR","avty_t":"tele/tasmota_B3F48A/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"B3F48A_AM2301_Temperature","dev":{"ids":["B3F48A"]},"unit_of_meas":"°C","dev_cla":"temperature","frc_upd":true,"val_tpl":"{{value_json['AM2301']['Temperature']}}"}
homeassistant/sensor/B3F48A_AM2301_Humidity/config {"name":"wTasmota01 AM2301 Humidity","stat_t":"tele/tasmota_B3F48A/SENSOR","avty_t":"tele/tasmota_B3F48A/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"B3F48A_AM2301_Humidity","dev":{"ids":["B3F48A"]},"unit_of_meas":"%","dev_cla":"humidity","frc_upd":true,"val_tpl":"{{value_json['AM2301']['Humidity']}}"}
homeassistant/sensor/B3F48A_AM2301_DewPoint/config {"name":"wTasmota01 AM2301 DewPoint","stat_t":"tele/tasmota_B3F48A/SENSOR","avty_t":"tele/tasmota_B3F48A/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"B3F48A_AM2301_DewPoint","dev":{"ids":["B3F48A"]},"unit_of_meas":"°C","dev_cla":"temperature","frc_upd":true,"val_tpl":"{{value_json['AM2301']['DewPoint']}}"}
homeassistant/sensor/B3F48A_status/config {"name":"wTasmota01 status","stat_t":"tele/tasmota_B3F48A/HASS_STATE","avty_t":"tele/tasmota_B3F48A/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"tele/tasmota_B3F48A/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"B3F48A_status","dev":{"ids":["B3F48A"],"name":"wTasmota01","mdl":"Generic","sw":"10.1.0(sensors)","mf":"Tasmota"}}
```

Maybe it would be possible to implement for this library as well? It would be really useful for those, who make similar sensors.

camnewnham, 2023-01-16:
I've just run into this as well and I also tried something similar your pointer solution. I was attempting to append a unique Id (MAC/ChipId) to the IDs.

In my case, I was using a `String` in the constructor that was going out of the `setup` scope and the memory seemed to being reassigned. If this is the case for you, you can see this by plotting `co2Ptr->uniqueId()` in the main loop. On the first plot it will (probably) appear correct but eventually become jibberish.

My fix was to define the ID string in global scope. Snippet:

```C++

WiFiClient wifiClient;   // MQTT Controller

// HA initialized in setup() to use chip-unique IDs
HADevice* haDevice;
HANumber* haSpeed;
HAMqtt* mqtt;

String speedId("coverboard_speed-");
String deviceId("coverboard-");

void setup() {
<...>
    String macStr =  String(mac[0],HEX) +String(mac[1],HEX) +String(mac[2],HEX) +String(mac[3],HEX) + String(mac[4],HEX) + String(mac[5],HEX);  
    deviceId.concat(macStr);
    speedId.concat(macStr);

    haDevice = new HADevice(deviceId.c_str());
    haDevice->setName("MyBoard");

    haSpeed = new HANumber(speedId.c_str(), HABaseDeviceType::NumberPrecision::PrecisionP2);

<...>
    mqtt = new HAMqtt(wifiClient, *haDevice);
    mqtt->begin(MQTT_SERVER, 1883U, MQTT_USER, MQTT_PASS);
}

```

simie7, 2023-01-17:
@camnewnham thanks for idea to print `uniqueId` - understood where was the problem. So for now this is working example:
```
HASensorNumber *co2Ptr = nullptr;

uint32_t chipId;                                    // For storing ESP chip ID
char co2Id[40], co2Name[40];

void setup() {
// <...>

  chipId = ESP.getChipId();

  sprintf(co2Id,   "%08X_T6603_CO2", chipId);
  sprintf(co2Name, "%08X T6603 CO2", chipId);
  co2Ptr = new HASensorNumber(co2Id);
  co2Ptr->setName(co2Name);
  co2Ptr->setUnitOfMeasurement("ppm");
  co2Ptr->setDeviceClass("carbon_dioxide");

// <...>
}

void loop() {

  mqtt.loop();
  co2Ptr->setValue(readCO2());

}
```
It registers and updates correctly. Not very nice way, but it works and is ready for templating.

justind000, 2023-03-05:
To add to the code @simie7 provided, for each sensor, you must have a dedicated char buffer for each sensor you add.

`char co2Id[40], co2Name[40];
char other_sensor_Id[40], other_sensor_name[40];`

Reusing a single char buffer for several sensors will result in only the last one being present on the MQTT broker. I assume the library stores the buffer locations and assembles the strings when needed, not when they are created.