dawidchyrzynski/arduino-home-assistant #129: Cant create more then 4 switches

EmilEmilchen, 2023-01-14:
For some reason, maybe even baucuse of Homeassistant or the MQTT broker Iam using, I can only create four switches. Iam using the following Code:
`
#include <Ethernet.h>
#include <ArduinoHA.h>

byte mac[] = {0x00, 0x10, 0xFA, 0x6E, 0x38, 0x4A};

#define device_name "Arduino tst2"
#define device_version "1.0.0"

const uint8_t pins[] = {13, 22, 24, 26, 28, 30, 32, 34, 36};
const String pin_assignments[] = {"int_led", "rel1", "rel2", "rel3", "rel4", "rel5", "rel6", "rel7", "rel8"};
const String names[] = {"INTERNAL LED", "RELAY 1", "RELAY 2", "RELAY 3", "RELAY 4", "RELAY 5", "RELAY 6", "RELAY 7", "RELAY 8"};

const int pin_count = sizeof(pins)/sizeof(pins[0]);
EthernetClient client;
HADevice device(mac, sizeof(mac));
HAMqtt mqtt(client, device);
HASwitch* led[2];

int CustomIndexOf(String to_find) {
    for (uint8_t i = 0; i < pin_count; i++) {
        if (pin_assignments[i] == to_find) {
            return i;
        }
    }   return -1;
}

void onSwitchCommand(bool curr_state, HASwitch* sender) {
    bool state = curr_state;
    digitalWrite(pins[CustomIndexOf(sender->uniqueId())], (state ? HIGH : LOW));
    sender->setState(state); // report state back to the Home Assistant
    Serial.println(CustomIndexOf(sender->uniqueId()));
}

void setup() {
    // start serial (for debugging)
    Serial.begin(9600);
    Serial.println("Starting...");
    Serial.println(sizeof(pins)/sizeof(pins[0]));
    // create switches
    for (uint8_t i = 0; i < pin_count; i++) {
        led[i] = new HASwitch(pin_assignments[i].c_str());
        Serial.println("Created led["+String(i)+"]: "+String(pin_assignments[i]));
        led[i]->setName(names[i].c_str());
        Serial.println("Name led["+String(i)+"]: "+String(names[i]));
        led[i]->onCommand(onSwitchCommand);
    }

    // set pin modes
    for (uint8_t i = 0; i < pin_count; i++) {
        pinMode(pins[i], OUTPUT);
        digitalWrite(pins[i], LOW);
    }

    //initiate the ethernet connection
    Ethernet.begin(mac);

    // set device's details
    device.setName(device_name);
    device.setSoftwareVersion(device_version);

    mqtt.begin("192.168.1.74", 1883, "homeassistant", "Ci9que4phoi8lootiZaek9ahchohzah1ri7ouKahl6oexe2Iegaijaevohniotae");
}

void loop() {
    Ethernet.maintain();
    mqtt.loop();
}
`
Only the first 4 entitys show up in homeassistat and can be used without a problem. I even tried Doing it manually and copy-pasting all the HASwitch'es and creating a seperate variable for all of them. I can still only get four of them to show up in home assistant.
I really dont know what to do!

EmilEmilchen, 2023-01-14:
I noticed that the first, second, third and fifth one work, the fourth one doesnt.

tomakara, 2023-01-16:
I have same problem. Only first five switches are registred on MQTT.

`
  HASwitch swi1("45");
  HASwitch swi2("55");
  HASwitch swi3("56");
  HASwitch swi4("57");
  HASwitch swi5("58");
  HASwitch swi6("59");
  HASwitch swi7("64");
  HASwitch swi8("67");

HASwitch sw[] = {
  HASwitch(RFsw[1][0]),
  HASwitch(RFsw[2][0]),
  HASwitch(RFsw[3][0]),
  HASwitch(RFsw[4][0]),
  HASwitch(RFsw[5][0]),
  HASwitch(RFsw[6][0]),
  HASwitch(RFsw[7][0]),
  HASwitch(RFsw[8][0]),
  HASwitch(RFsw[9][0]),
  HASwitch(RFsw[10][0]),
  HASwitch(RFsw[11][0]),
  HASwitch(RFsw[12][0])
};

`

### MQTT Explorer:
_

homeassistant
switch
lights_basement
45
config = {"uniq_id":"45","dev":{"ids":"lights_basement","name":"Lights Relay switch basement","mf":"VilStein","mdl":"LRS1","sw":"0.1.2"},"stat_t":"aha/lights_basement/45/stat_t","cmd_t":"aha/lights_basement/45/cmd_t"}
55
config = {"uniq_id":"55","dev":{"ids":"lights_basement","name":"Lights Relay switch basement","mf":"VilStein","mdl":"LRS1","sw":"0.1.2"},"stat_t":"aha/lights_basement/55/stat_t","cmd_t":"aha/lights_basement/55/cmd_t"}
56
config = {"uniq_id":"56","dev":{"ids":"lights_basement","name":"Lights Relay switch basement","mf":"VilStein","mdl":"LRS1","sw":"0.1.2"},"stat_t":"aha/lights_basement/56/stat_t","cmd_t":"aha/lights_basement/56/cmd_t"}
57
config = {"uniq_id":"57","dev":{"ids":"lights_basement","name":"Lights Relay switch basement","mf":"VilStein","mdl":"LRS1","sw":"0.1.2"},"stat_t":"aha/lights_basement/57/stat_t","cmd_t":"aha/lights_basement/57/cmd_t"}
58
config = {"uniq_id":"58","dev":{"ids":"lights_basement","name":"Lights Relay switch basement","mf":"VilStein","mdl":"LRS1","sw":"0.1.2"},"stat_t":"aha/lights_basement/58/stat_t","cmd_t":"aha/lights_basement/58/cmd_t"}
aha
lights_basement
45
stat_t = OFF
55
stat_t = OFF
56
stat_t = OFF
57
stat_t = OFF
58
stat_t = OFF
_

Drakus66, 2023-01-16:
Have the same problem! I have 6 sensors and 6 HANumbers in Mega. But only first 5 entities registring in MQTT, no matter sensors or numbers they are

Drakus66, 2023-01-17:
I believe this is the solution
https://dawidchyrzynski.github.io/arduino-home-assistant/documents/library/device-types.html#

will try it later

tomakara, 2023-01-17:
> I think this is the solution https://dawidchyrzynski.github.io/arduino-home-assistant/documents/library/device-types.html#
> 
> will try it later

It is the solution.
Thank you!

closed 2023-01-23 by dawidchyrzynski