dawidchyrzynski/arduino-home-assistant #131: light.setState publishing non-retained state

milansim, 2023-01-18:
light.setState works, but after arduino restart state is not saved, despite setretain(true), only if turned on/off from Home assistant

My workaround is complex...I use tzapu's wifimanager to save topic for lights (e.g. "aha/c8f09e52f9d4/")
then instead of set state I publish on cmd/t
String light1 = light_topic + "Light1/cmd_t";
mqtt.publish(light1.c_str(), "ON", true); // mqtt.publish("customTopic", "customPayload", true); //true mean retained

Obviously "setState" needs retained flag in library like it has "force" flag

dawidchyrzynski, 2024-02-05:
@milansim Closing this as I am unable to reproduce the issue. Retention works correctly in both directions; hence, it might be a bug within your implementation.

closed 2024-02-05 by dawidchyrzynski

milansim, 2024-02-08:
> @milansim Closing this as I am unable to reproduce the issue. Retention works correctly in both directions; hence, it might be a bug within your implementation.

It is simple implementation so I doubt that. I will test again sometime but since my workaround works...

amalon, 2024-10-22:
I see something similar. If I set rgb from HA, it puts a rgb_cmd_t in mqtt, but if I then send a button press to the device to trigger a reset to default colours, i have it changing state unilaterally from the arduino device, that changes rgb_stat_t, but leaves rgb_cmd_t at the now stale value set by HA, which is reloaded on reset.