dawidchyrzynski/arduino-home-assistant #137: Support Arduino DUE?

proasnet, 2023-02-13:
Hello, my project I want migrate from Mega to Due,
compilation return errors for
setValue

is possible, please, support Due board?
Thanks

dawidchyrzynski, 2023-02-20:
Hi @proasnet,

Can you provide the build's log?

proasnet, 2023-02-22:
Hi @dawidchyrzynski 

This is compile log
```
:\Jirutka\99_VYVOJ ARDUINO\Axpert\AxpertinoGW_V1.11_DUE\AxpertinoGW_V1.11_DUE.ino: In function 'void loop()':
AxpertinoGW_V1.11_DUE:735:69: error: call of overloaded 'setValue(int&)' is ambiguous
     Inv_1_outputPowerApparent.setValue( hodnoty_QPIGS_Inv_1.AC_O_VA );
                                                                     ^
Z:\Jirutka\99_VYVOJ ARDUINO\Axpert\AxpertinoGW_V1.11_DUE\AxpertinoGW_V1.11_DUE.ino:735:69: note: candidates are:
In file included from C:\Users\j.jirutka\Documents\Arduino\libraries\home-assistant-integration\src/ArduinoHA.h:11:0,
                 from Z:\Jirutka\99_VYVOJ ARDUINO\Axpert\AxpertinoGW_V1.11_DUE\AxpertinoGW_V1.11_DUE.ino:10:
C:\Users\j.jirutka\Documents\Arduino\libraries\home-assistant-integration\src/device-types/HASensor.h:37:10: note: bool HASensor::setValue(const char*) <near match>
     bool setValue(const char* value);
          ^
C:\Users\j.jirutka\Documents\Arduino\libraries\home-assistant-integration\src/device-types/HASensor.h:37:10: note:   no known conversion for argument 1 from 'int' to 'const char*'
C:\Users\j.jirutka\Documents\Arduino\libraries\home-assistant-integration\src/device-types/HASensor.h:38:10: note: bool HASensor::setValue(uint32_t)
     bool setValue(uint32_t value);
          ^
C:\Users\j.jirutka\Documents\Arduino\libraries\home-assistant-integration\src/device-types/HASensor.h:39:10: note: bool HASensor::setValue(int32_t)
     bool setValue(int32_t value);
```

etc..

dawidchyrzynski, 2024-02-09:
The Arduino Due support will be added in the upcoming version of library.

closed 2024-02-09 by dawidchyrzynski