dawidchyrzynski/arduino-home-assistant #138: MQTT Payload always contains HA Config

Blueforcer, 2023-02-20:
besides HA i use the libs MQTT cleint also for other things.
if i receive a message, there is always some ha config in the payload.
in following example i only send "hallo"

hallo1097bd33921c/BTN8/config���␄�B���>5E��;_w��EY�f��R␑�␎6�&��␛�R�W��ԙb���p�ė�I␗�␎&l*�����J꣨:mb␝���W␝:l�^A���␗]۪�S�{�[�����h␇�ە��_�s␜�␗�␆{|����~��␚�␡2l␚H�^��*'␑�m���␘�Fӟ;>��,��␅�Lg2̹@�>���7�j���mrG�q_��^ A�␎=^�0�t��GUjIxV��T

i cannot find my error:)

dawidchyrzynski, 2023-02-20:
Hi @Blueforcer,

Please provide the source code.

Blueforcer, 2023-02-20:
Hi @dawidchyrzynski 

there is nothing wich could change my payload
[image]

Output is
[image]

you can find my sourcecode here:
https://github.com/Blueforcer/SmartPusher/blob/17aa11c9569d6068e79de34a6582efa5960752d0/src/MqttManager.cpp#L90

dawidchyrzynski, 2023-02-20:
The `onMessage` callback function is designed to handle both external and internal messages that are processed by the library. However, if the message topic does not meet your requirements, it should be ignored. Therefore, it is important to implement the necessary checks to filter out irrelevant messages. It's expected behaviour. See the documentation: https://dawidchyrzynski.github.io/arduino-home-assistant/documents/api/core/ha-mqtt.html

closed 2023-02-20 by dawidchyrzynski

Blueforcer, 2023-02-20:
I already filter the topics.
While the topic is correct, the payload is still mixed up, it contains my sended message AND some internal messages.
[image]

Blueforcer, 2023-02-20:
I already filter my topics.
While the topic is correct the payload is still messed up. It contains my message (green box) AND some internal data
[image]

dawidchyrzynski, 2023-02-20:
@Blueforcer That's due to an invalid conversion of uint8_t to String. The payload buffer doesn't contain a null terminator.