dawidchyrzynski/arduino-home-assistant #140: Mqtt.loop blocking

PM04290, 2023-02-21:
If no mqtt server available, or wifi/ethernet fail, mqtt.loop() freeze the main loop during many seconds.
is it possible to modify process of mqtt.loop with non blocking ? something like true/false return.
i must fork this library to solve problem and it's not a good solution for future.

are you knowing "[async-mqtt-client](https://github.com/marvinroger/async-mqtt-client)" library ? is it possible to choose this one or PubSubClient ?

xardas689, 2023-03-15:
The async-mqtt-client library is only working on esp devices. This library is designed to work with avr based processors as well and therfore using the PubSubClient internally with a wrapper build on top of it.

Besides that: I am not able to reproduce the error on an arduino mega board. But i noticed that the wrapper is calling the underlying loop function directly. It may help if you use the [nonblocking reconnection](https://github.com/knolleary/pubsubclient/blob/master/examples/mqtt_reconnect_nonblocking/mqtt_reconnect_nonblocking.ino).

rikki78, 2023-06-25:
the nonblocking reconnection still will block the loop, as it waits 15s until it gets connected. If you want to avoid this you have to run the entire mqtt.loop(); in a different ESP task than the program loop

dawidchyrzynski, 2024-01-31:
Duplicate of #77

closed 2024-01-31 by dawidchyrzynski