dawidchyrzynski/arduino-home-assistant #142: Add effects list

mlc76520, 2023-02-26:
First thanks for this and the hard work. This library is really amazing.
I'd like to have the opportunity to get a list of effects in home assistant mapped to some custom effects available on the arduino side. ie a rainbow effect.

Hope you will consider the request if not too challenging,

Best regards,

addyvandendoel, 2023-03-03:
Add me to this please! 
I also Like to be able to set effects for the light. Used the library for connecting a color ledstrip WS2813 to and esp8266. Getting it lit up with this is amazing, just missing out the effects option now. 
 I guess I could fix this myself with the onMessage( ) function, but would be really nice to have an effect list feature for the light.

martaisty, 2023-05-21:
Plus 1 on this. :+1:

martaisty, 2023-07-23:
@mlc76520 , @addyvandendoel  Hello :wave:  It's been a while since the issue was created :smile: I also needed this feature and implemented it in PR #178. Until it's merged, feel free to use changes on my fork https://github.com/martaisty/arduino-home-assistant/tree/feature/light-effects-list

martaisty, 2024-02-10:
Hi @dawidchyrzynski 👋 I see that you are actively doing changes on the project. I was wondering if there is a chance to get this issue resolved. I will resolve the conflicts on my PR and kindly ask to look at it 😃

dawidchyrzynski, 2024-02-10:
Hi @martaisty, no problem. Let's aim to merge it asap.