dawidchyrzynski/arduino-home-assistant #143: ESP32 Support for Method Pointers

ZanzyTHEbar, 2023-02-26:
Hello, i am using this library, and i am having to re-write the call backs to support `std::function` on the ESP devices. 

In my project, i have created a `HASSMQTT` class for handling all of my HAMqtt logic. Normally, i would use a lambda or `std::bind` to a callback which has a `std::function` based type. 

I would really like this functionality - but i can't in the default library as you only except function points. 

I would recommend doing something like:
````cpp
#ifdef ESP32
typedef std::function<void(bool state, HASwitch* s)> callback;
#else
#ifdef ESP8266
typedef std::function<void(bool state, HASwitch* s)> callback;
#endif
void (*callback)(bool, HASwitch *);
#endif
```

Now, of course the `std::function` is available for all 32bit boards, so adding support for all 32bit boards using a single preprocessor macro is preferred.

KSDaemon, 2024-12-07:
Yeah, exactly the same situation.

KSDaemon, 2024-12-07:
A bit shorter version of @ZanzyTHEbar Suggestion: 
```cpp
#if defined(ESP32) || defined(ESP8266)
#define HASWITCH_CALLBACK(name) std::function<void(bool state, HASwitch* s)> name
#else
#define HASWITCH_CALLBACK(name) void (*name)(bool state, HASwitch* sender)
#endif
```

PR #281 (2025-07-06): Add support for std::function on the ESP devices
Hi! Merry Christmas and Happy New Year!

This PR adds conditional compilation flow for ESP32/ESP8266 platforms to allow using the `std::function` as a command callback.

I hope I've found all the places. But if it's not true — please point me and I'll update the missing places.

This closes #143.
closed 2025-07-07 by dawidchyrzynski