dawidchyrzynski/arduino-home-assistant #146: HASelect.setState() should allow setting -1

rrozema, 2023-03-05:
It would be very helpful to me if HASelect would allow to set the state back to -1 after it has been some valid state. In HA this should result in an update to the same 'unknown' value that is shown initially when no state has been assigned yet.

My scenario is a fake "traffic light", where the 3 lights can be controlled by individual switches per light but the lights can also be controlled via a "traffic light" select. If the individual switches are in a valid "traffic light"-combination (exactly 1 light is on), the "traffic light" select should show this combination's name. When however an invalid combination is chosen (0, 2 or 3 lights are on at the same time) using the switches, the select should be updated to the -1 - 'unknown' value.

rrozema, 2023-03-14:
I've been testing this and I don't know if it's even possible to set a value that is not in the list of options (i.e. this is what would be needed to set -1). So far I haven't been able to do it myself, so I won't blame you if you reject this ;-) , but if you could have a look to see if you can think of a way to (re-)set the select's state -1 and have HA show 'unavailable' (or something like that) I would certainly appreciate it!

dawidchyrzynski, 2024-02-05:
This feature will be available in the upcoming version of the library. Please refer to the linked Pull Request for more information.

closed 2024-02-05 by dawidchyrzynski