dawidchyrzynski/arduino-home-assistant #147: sensor value not updating in HA

rozrabiak, 2023-03-06:
Hi, why I can't send in below code IP address to HASensor? Always in dashboard I have "unknown" value..

HADevice haDevice(mac, sizeof(mac));
HAMqtt hamMqtt(wifiClient, haDevice, 20);

HASensor haIpAddress("adresIp");

void setup()
{
if (hamMqtt.begin(BROKER_ADDR,"USER","PASS"))    
    Serial.println("Mqtt connected");
  delay(2000);

  haIpAddress.setName("Adres IP");
  char ip[] = "xxx.xxx.xxx.xxx";
  IPAddress ips = WiFi.localIP();
  ips.toString().toCharArray(ip, 16);
  haIpAddress.setValue(ip); //wyslanie ip adresu do HA
}

void loop()
{
hamMqtt.loop();
}

AHA: connecting, client ID: 0010fa6e384a
AHA: connected
  
AHA: begin publish homeassistant/sensor/0010fa6e384a/tempOdczyt/config, len: 200
AHA: begin publish homeassistant/sensor/0010fa6e384a/czasSterownika/config, len: 174
AHA: begin publish homeassistant/sensor/0010fa6e384a/nastStartPompy/config, len: 181
AHA: begin publish homeassistant/sensor/0010fa6e384a/adresIp/config, len: 161
AHA: begin publish homeassistant/sensor/0010fa6e384a/mocSygnalu/config, len: 183
AHA: begin publish homeassistant/switch/0010fa6e384a/btnPompa/config, len: 232
AHA: subscribing aha/0010fa6e384a/btnPompa/cmd_t
AHA: begin publish homeassistant/switch/0010fa6e384a/onOff/config, len: 222
AHA: subscribing aha/0010fa6e384a/onOff/cmd_t
AHA: begin publish homeassistant/number/0010fa6e384a/tempZadana/config, len: 295
AHA: subscribing aha/0010fa6e384a/tempZadana/cmd_t
AHA: begin publish homeassistant/number/0010fa6e384a/interwal/config, len: 272
AHA: subscribing aha/0010fa6e384a/interwal/cmd_t
AHA: begin publish homeassistant/number/0010fa6e384a/godzinaRano/config, len: 285
AHA: subscribing aha/0010fa6e384a/godzinaRano/cmd_t
AHA: begin publish homeassistant/number/0010fa6e384a/minutyRano/config, len: 281
AHA: subscribing aha/0010fa6e384a/minutyRano/cmd_t
AHA: begin publish homeassistant/number/0010fa6e384a/godzinaNoc/config, len: 281
AHA: subscribing aha/0010fa6e384a/godzinaNoc/cmd_t
AHA: begin publish homeassistant/number/0010fa6e384a/deltaTemp/config, len: 276
AHA: subscribing aha/0010fa6e384a/deltaTemp/cmd_t
AHA: begin publish aha/0010fa6e384a/tempOdczyt/stat_t, len: 2

rozrabiak, 2023-03-06:
In HomeAssistant

[image]

dawidchyrzynski, 2023-03-06:
Hi @rozrabiak,

Sending value to HA can only be performed when the MQTT connection has been established. Therefore, it is recommended that you move the `haIpAddress.setValue(...)` call to the `HAMqtt::onConnected()` callback, which is triggered once the connection has been established.

https://dawidchyrzynski.github.io/arduino-home-assistant/documents/library/mqtt-advanced.html

closed 2023-03-06 by dawidchyrzynski