dawidchyrzynski/arduino-home-assistant #152: Stop command

joeyjojo747, 2023-03-25:
I am using the cover to control a stepper motor.
The stepper motor control uses a while loop to run the motor until a limit switch is activated.
Im having trouble incorporating the HACover::CommandStop.
Once the open command is received it stays in the while loop so the stop command isn't received until the loop is completed.
Help appreciated.
void onCoverCommand(HACover::CoverCommand cmd) {
  if (cmd == HACover::CommandOpen) {
    Serial.println("Command: Open");
    WebSerial.println("Command: Open");
    digitalWrite(stepperPin_sleep, LOW); //wake up the motor driver
    digitalWrite(dirPin, LOW); //Changes the rotations direction
    while (digitalRead(doorOpenLimit) == HIGH) {
      digitalWrite(stepPin, HIGH);
      delayMicroseconds(speed);
      digitalWrite(stepPin, LOW);
      delayMicroseconds(speed);
      yield();
    }
    digitalWrite(stepperPin_sleep, HIGH); // put the motor driver to sleep
    setDoorState();
    //cover.setState(HACover::StateOpening);
    
  } else if (cmd == HACover::CommandClose) {
    Serial.println("Command: Close");
    WebSerial.println("Command: Close");
    closeDoorCommand();
    //cover.setState(HACover::StateClosing);
    
  } else if (cmd == HACover::CommandStop) {
    Serial.println("Command: Stop");
    WebSerial.println("Command: Stop");
    //cover.setState(HACover::StateStopped);
    digitalWrite(stepperPin_sleep, HIGH);
  }

  // Available states:
  // HACover::StateClosed
  // HACover::StateClosing
  // HACover::StateOpen
  // HACover::StateOpening
  // HACover::StateStopped

  // You can also report position using setPosition() method
}
[doggy_door_v3_rebuild1_mqttDiscovery.zip](https://github.com/dawidchyrzynski/arduino-home-assistant/files/11068472/doggy_door_v3_rebuild1_mqttDiscovery.zip)

rrozema, 2023-03-25:
Have a look at non-blocking calls using the millis() function. For an example see the [BlinkWithoutDelay](https://docs.arduino.cc/built-in-examples/digital/BlinkWithoutDelay) example. 

The idea is that you do not call while nor delay to control the motor in a loop once you've received a start command, but instead take it one step at a time: every time the loop() function gets called you check to see if it is time yet for a next action. If it isn't time yet, you can do other things like for example checking to see if someone sent you a stop command.

joeyjojo747, 2023-04-03:
I got the non-blocking method working.
Thank you for your guidance.

closed 2023-04-03 by joeyjojo747