dawidchyrzynski/arduino-home-assistant #161: One physical device, need multiple HA devices each with multiple sensors

psgcooldog, 2023-04-14:
I'm creating an ESP32-based LORA gateway for multiple physical sensors.  I already have it working as one HA "Device" with multiple sensors, but would like to have each of the physical sensors (with their multiple HA sensor entities) appear as a separate HA Device.

I can't see any way to do this, as it appears that the first HADevice that is created "owns" all HADevices that are created subsequently.

Can you advise if this is even possible?

Thanks for the beautiful lib!

chinswain, 2024-05-19:
I have the same requirement - 10 beehives all transmitting via LoRa to a receiver which then publishes the devices to HA, having each hive as a separate HA device would be good.

psgcooldog, 2024-05-20:
I hacked away at it until I made it work the way I wanted it to.

chinswain, 2024-05-20:
Did you manage to add multiple devices or have you had to put them all as sensors on the one?

I've got Beehive1Temp, Beehive2Temp etc but it's really messy and lengthy code and looks a complete mess at the other end in HA with so many sensors!

chinswain, 2024-05-21:
Yea please, it would be interesting to take a look, I'm nearly at 100 sensors (which should be spread across 10 devices)

psgcooldog, 2024-05-21:
Can you send me your email address, or is there some sort of messaging
capability here?  I haven't touched this stuff in probably 18 months or
more.  The best thing would be to send you the entire PlatformIO project,
which includes the ESP32 gateway, and the sensor end for a Heltec CubeCell.