dawidchyrzynski/arduino-home-assistant #164: HASelect doesn't work

netmindz, 2023-05-04:
The example select.ino had issues with select being redefined but if swap to selectTest, I then see the device appear in HA, but changes to HA don't call onSelectCommand nor does calling setState actually appear to trigger a change in HA

netmindz, 2023-05-04:
ok, worked for a moment, now broken again

variable name should still be changed in the example

netmindz, 2023-05-04:
Think this new version is having mqtt issues. It says connected if I attach a onConnected method, but updates not coming into HA. Swapping back to 1.3.0 resolves the issue.

Is there a way to get debug logs from the library?

dawidchyrzynski, 2024-02-05:
@netmindz Closing this as I'm unable to reproduce the issue. You can enable the debug logs by uncommenting the `// #define ARDUINOHA_DEBUG` line in the `ArduinoHADefines.h` file.

closed 2024-02-05 by dawidchyrzynski

Xieo, 2024-06-25:
Hi Guys, I found out that when the options set as numbers     Dropdown.setOptions("0;1;2");

the code works and it calls the onSelectCommand().

but when set at text Dropdown.setOptions("Cool;Hot;Off");  it doesn't trigger the onSelectCommand()  and caused the mqtt to reset the connection.

any recommendation?