dawidchyrzynski/arduino-home-assistant #166: Esp8266/Esp12f does not work,restart loop

yukiloh, 2023-05-20:
Fully copied example code
https://github.com/dawidchyrzynski/arduino-home-assistant/blob/main/examples/nodemcu/nodemcu.ino

Just modified the WiFi info
```
#define WIFI_SSID       "my wifi"
#define WIFI_PASSWORD   "my password"
```
When it run,error like:
```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4000bf80 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd50 end: 3fffffc0 offset: 0190
3ffffee0:  3ffee588 0000000d 4023da94 40201b85  
3ffffef0:  000010ed 7fffffff 000000a0 3fffff20  
3fffff00:  3fffdad0 3ffefc34 3ffee7c0 3fffff20  
3fffff10:  3fffdad0 3ffee7e8 3ffee7c0 4020274a  
3fffff20:  514d0400 5a045454 4bc6a701 00000000  
3fffff30:  00000001 3ffefaf4 00000000 3ffee9dc  
3fffff40:  3fffdad0 3ffee7e8 3ffee7c0 40201acf  
3fffff50:  3fffdad0 3ffee7e8 00000001 402018ad  
3fffff60:  3fffdad0 00000000 3ffee7e8 4020191d  
3fffff70:  00000000 00000000 00000001 3ffee9dc  
3fffff80:  3fffdad0 00000000 3ffee7e8 40201944  
3fffff90:  40208dd0 5a01a8c0 3ffee9c8 402014c4  
3fffffa0:  3fffdad0 00000000 3ffee9c8 40205360  
3fffffb0:  feefeffe feefeffe 3ffe85d8 40100d31  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

Is this lib not support Esp8266/Esp12F?
But I tested in my Esp32,it's working.

SimonRafferty, 2023-05-21:
I'm having the same issue, running on an Adafruit Huzzah32 (ESP32).
It reboots on the line: analogSensor.setValue(voltage);
From the Analog Sensor example.

dawidchyrzynski, 2024-02-05:
I'm unable to reproduce this issue. Could you please tell me which version of the ESP32/ESP8266 board manager you are using?

dawidchyrzynski, 2025-07-07:
This issue will be fixed in the coming version of the library - #292

closed 2025-07-07 by dawidchyrzynski