dawidchyrzynski/arduino-home-assistant #167: HAButton that trigger service on HomeAssistant

mariusmotea, 2023-05-21:
Hi,

I have an Elecrow ESP32-S3 display with touch control and i want to have it like a room dashboard. I also want to have some buttons on the display that will control devices connected to HomeAssistant (ex: garage door, vacuum robot) using some logic at home assistant level. Currently to achieve this i need to create a switch that is toggled on/off with a delay in-between, but according to home assistant this may also be possible with MQTT button integration?

> The mqtt button platform lets you send an MQTT message when the button is pressed in the frontend or the button press service is called

erkr, 2023-11-27:
I think the HADeviceTrigger is what you need for that

dawidchyrzynski, 2024-01-30:
https://github.com/dawidchyrzynski/arduino-home-assistant/blob/main/examples/multi-state-button/multi-state-button.ino

closed 2024-01-30 by dawidchyrzynski

mariusmotea, 2024-02-18:
Unfortunately i tried entire day the multi state button example and nothing is happening in Home Assistant. I see the actions in the MQTT broker with MQTT Explorer, but HomeAssistant ignore the events. HAButton and HASwitch are working fine if i set such devices. How i can link the long press and short press to the same entity? I have some wall switches with 3 buttons and i want to get in HomeAssistant one button entity with the type button_short_press and button_long_press and the subtype button_1, button_2 and button_3.

Thanks.