dawidchyrzynski/arduino-home-assistant #172: 4th and 6th switch doesnt work!

EmilEmilchen, 2023-06-11:
Hi.
I'am using an Arduino Mega with a relay board with 8 relays and I'm trying to control the 8 relays as well as the Arduinos internal LED using MQTT and Home Assistant. The Problem is that only the internal LED and 6 out of the 8 relays work properly. The other two relays are initially beeing reported to the MQTT Broker by only their id and not their friendly name but then fail to work. triggering the relays using either the home assistant gui or a mqtt broker doesnt cause anything to happen. The command is getting send in the .../cmd_t topic but the command either doesnt even reach the Arduino or isnt registered by it. I assume the latter because the name isnt reported to the mqtt broker. The two relays which dont work are also always "RELAY 3" and "RELAY 5", at least in this example. It doesnt matter if you switch "RELAY 2" and "RELAY 3" (and the according pin numbers and ids), the problem always only regards the 4th and 6th switch ("entity"). I have absolutely no clue why that is!
Here what is initially send in the confg topic by the Arduino:
`{"uniq_id":"rel3","dev":{"ids":"0010fa6e384a","name":"Arduino tst2","sw":"1.0.0"},"stat_t":"aha/0010fa6e384a/rel3/stat_t","cmd_t":"aha/0010fa6e384a/rel3/cmd_t"}`
The state is also reported once. Initially, when the config also gets send, the Arduino reports "OFF".
Below you can find my Code and even further down I also put the diagnostic Information from Home Assistant which you can download from the GUI.

I really help anyone will be able to help me. I pretty much tried everything I could think of.

My Code:
`#include <Ethernet.h>
#include <ArduinoHA.h>

byte mac[] = {0x00, 0x10, 0xFA, 0x6E, 0x38, 0x4A};

#define device_name "Arduino test 2"
#define device_version "1.0.0"

const uint8_t pins[] = {13, 22, 24, 26, 28, 30, 32, 34, 36};
const String pin_assignments[] = {"int_led", "rel1", "rel2", "rel3", "rel4", "rel5", "rel6", "rel7", "rel8"};
const String names[] = {"INTERNAL LED", "RELAY 1", "RELAY 2", "RELAY 3", "RELAY 4", "RELAY 5", "RELAY 6", "RELAY 7", "RELAY 8"};

const int pin_count = sizeof(pins)/sizeof(pins[0]);
EthernetClient client;
HADevice device(mac, sizeof(mac));
HAMqtt mqtt(client, device, 20);
HASwitch* led[2];

int CustomIndexOf(String to_find) {
    for (uint8_t i = 0; i < pin_count; i++) {
        if (pin_assignments[i] == to_find) {
            return i;
        }
    } return -1;
}

void onSwitchCommand(bool curr_state, HASwitch* sender) {
    bool state = curr_state;
    digitalWrite(pins[CustomIndexOf(sender->uniqueId())], (state ? HIGH : LOW));
    sender->setState(state); // report state back to the Home Assistant
    Serial.println(CustomIndexOf(sender->uniqueId()));
    Serial.println(state);
}

void setup() {
    // start serial (for debugging)
    Serial.begin(9600);
    Serial.println("Starting...");
    Serial.println(sizeof(pins)/sizeof(pins[0]));
    // create switches
    for (uint8_t i = 0; i < pin_count; i++) {
        led[i] = new HASwitch(pin_assignments[i].c_str());
        Serial.println("Created led["+String(i)+"]: "+String(pin_assignments[i]));
        led[i]->setName(names[i].c_str());
        Serial.println("Name led["+String(i)+"]: "+String(names[i]));
        led[i]->onCommand(onSwitchCommand);
    }

    // set pin modes
    for (uint8_t i = 0; i < pin_count; i++) {
        pinMode(pins[i], OUTPUT);
        digitalWrite(pins[i], LOW);
    }

    //initiate the ethernet connection
    if (Ethernet.begin(mac) == 0) {
        Serial.println("Failed to configure Ethernet using DHCP");
    }
    
    Serial.print("My IP address: ");
[... 869 lines cut ...]
                  "topic": "aha/0010fa6e384a/rel8/cmd_t"
                },
                {
                  "payload": "ON",
                  "qos": 0,
                  "retain": false,
                  "time": "2023-06-11T15:30:07.024611+00:00",
                  "topic": "aha/0010fa6e384a/rel8/cmd_t"
                },
                {
                  "payload": "OFF",
                  "qos": 0,
                  "retain": false,
                  "time": "2023-06-11T15:30:18.700464+00:00",
                  "topic": "aha/0010fa6e384a/rel8/cmd_t"
                }
              ]
            }
          ]
        }
      ],
      "triggers": []
    }
  }
}`

kamil-jakubowski, 2023-06-13:
I have a little different, but maybe similar problem.
Every my sixth defined switch/led in the code does not work.

```
HASwitch valveRotDom("valve_rot_dom");
HASwitch valveStatDom("valve_stat_dom");
HASwitch valvePlot("valve_dupa");
HASwitch valveRotTeren("valve_rot_teren");
HASwitch valveStatTeren("valve_stat_teren");
HASwitch lightOutSwitch("light_out");
```
```
function setup(): void {
// [...]
// set icon (optional)
    valveRotTeren.setIcon("mdi:pipe-valve");
    valveRotTeren.setName("Sekcja Rotacyjne Teren");

    valveRotDom.setIcon("mdi:pipe-valve");
    valveRotDom.setName("Sekcja Rotacyjne Dom");

    valveStatTeren.setIcon("mdi:pipe-valve");
    valveStatTeren.setName("Sekcja Statyczne Teren");

    valveStatDom.setIcon("mdi:pipe-valve");
    valveStatDom.setName("Sekcja Statyczne Dom");

    valvePlot.setIcon("mdi:pipe-valve");
    valvePlot.setName("Sekcja Dupa");

    lightOutSwitch.setName("Swiatlo ogrod");

    // handle switch state
    lightOutSwitch.onCommand(switchLight);

    valveRotTeren.onCommand(switchRotTeren);
    valveRotDom.onCommand(switchRotDom);
    valveStatTeren.onCommand(switchStatTeren);
    valveStatDom.onCommand(switchStatDom);
    valvePlot.onCommand(switchBrama);
// [...]
}
In this case lightOurSwitch does not appear in mqtt autodiscovery. When I switch the order of defining objects:

```
HASwitch valveRotDom("valve_rot_dom");
HASwitch valveStatDom("valve_stat_dom");
HASwitch valvePlot("valve_dupa");
HASwitch valveRotTeren("valve_rot_teren");
HASwitch lightOutSwitch("light_out");
HASwitch valveStatTeren("valve_stat_teren");
```
`lightOutSwitch` is starting to be appear and working and `valveStatTeren` which have been already autodiscovered in HA is working fine in HA and MQTT (cmd_t is sent), but Arduino does not respond and does not even reach `onCommand` event.

When I turn the order back and have also `lightOuSwitch` registered in arduino - `valveStatTeren` in working again, and the light is starting to behave like every sixth instance - `cmd_t` goes to mqtt, but Arduino cannot respond on it.

shotreal, 2023-06-18:
I can also second this exact behavior. I have not had time to take a closer look at the library yet but my 6th device somehow is not recognized by HA either. Changing the order leads to the behavior described by [kamil-jakubowski](https://github.com/kamil-jakubowski). 

I had a brief look at the respective ha discovery topic and apparently there are no messages sent.
I have not yet had time to have a closer look at the code of the library, yet.

shotreal, 2023-06-18:
so, kids are sleeping; had a look at the code; number of devices is limited to 6 by default

from  HAMqtt.h
`    explicit HAMqtt(
        Client& netClient,
        HADevice& device,
        const uint8_t maxDevicesTypesNb = 6
    );
`

just specify the desired number of devices in the mqtt constructor e.g. ...
`HAMqtt mqtt(client, device,10);`

EmilEmilchen, 2023-06-18:
Thank you but my problem doesnt seem to be realted to the others in this Thread after all since I did spicify the maximum number of devices to be 20, so more then enough. But it still doesnt work. And the 4th and 6th switches wont work.

dawidchyrzynski, 2023-06-18:
Hi @EmilEmilchen, I will check this issue later in the upcoming week. I apologize for the inconvenience caused by the lack of support. Currently, I am occupied with other projects.

kamil-jakubowski, 2023-06-27:
@shotreal Thanks for the advice. I will try it next weekend! 👍

ChristofSchmid, 2023-06-29:
Hi
I have the same problem.
One device with 24 sensors. Only the first 5 sensors are detected. If neded I can share code and logs. 
Best, Christof

EmilEmilchen, 2023-06-30:
Hi @ChristofSchmid, this sounds like it's the same problem kamil-jakubowski had. I assume you did specify the maximum number of devices to be 24?

ChristofSchmid, 2023-06-30:
Hi Emil
Thanks for your reply.
I did not set the number of devices. Maybe a misunderstanding reading the documentation.
I have one device with a bunch of sensors(its even more than 24, we are at 40 ATM).
Physically, I run 8 humidity sensors which report soilHumidity, ambTemperature, batteryVoltage. We also need the timestamp of the reading and the “human readable” name of the sensor which can be entered in the web interface and saved to ESP32.

Here my setup:

WiFiClient wClient;
HADevice haDevice;
HAMqtt mqtt(wClient, haDevice);

HASensorNumber Sensor1("Sensor1", HASensorNumber::PrecisionP0);
HASensorNumber Sensor2("Sensor2", HASensorNumber::PrecisionP0);
HASensorNumber Sensor3("Sensor3", HASensorNumber::PrecisionP0);
HASensorNumber Sensor4("Sensor4", HASensorNumber::PrecisionP0);
HASensorNumber Sensor5("Sensor5", HASensorNumber::PrecisionP0);
HASensorNumber Sensor6("Sensor6", HASensorNumber::PrecisionP0);
HASensorNumber Sensor7("Sensor7", HASensorNumber::PrecisionP0);
HASensorNumber Sensor8("Sensor8", HASensorNumber::PrecisionP0);

HASensorNumber Time1("Time1", HASensorNumber::PrecisionP0);
HASensorNumber Time2("Time2", HASensorNumber::PrecisionP0);
HASensorNumber Time3("Time3", HASensorNumber::PrecisionP0);
HASensorNumber Time4("Time4", HASensorNumber::PrecisionP0);
HASensorNumber Time5("Time5", HASensorNumber::PrecisionP0);
HASensorNumber Time6("Time6", HASensorNumber::PrecisionP0);
HASensorNumber Time7("Time7", HASensorNumber::PrecisionP0);
HASensorNumber Time8("Time8", HASensorNumber::PrecisionP0);

HASensorNumber Soil1("Soil1", HASensorNumber::PrecisionP1);
HASensorNumber Soil2("Soil2", HASensorNumber::PrecisionP1);
HASensorNumber Soil3("Soil3", HASensorNumber::PrecisionP1);
HASensorNumber Soil4("Soil4", HASensorNumber::PrecisionP1);
HASensorNumber Soil5("Soil5", HASensorNumber::PrecisionP1);
HASensorNumber Soil6("Soil6", HASensorNumber::PrecisionP1);
HASensorNumber Soil7("Soil7", HASensorNumber::PrecisionP1);
HASensorNumber Soil8("Soil8", HASensorNumber::PrecisionP1);

HASensorNumber Temp1("Temp1", HASensorNumber::PrecisionP1);
HASensorNumber Temp2("Temp2", HASensorNumber::PrecisionP1);
HASensorNumber Temp3("Temp3", HASensorNumber::PrecisionP1);
HASensorNumber Temp4("Temp4", HASensorNumber::PrecisionP1);
HASensorNumber Temp5("Temp5", HASensorNumber::PrecisionP1);
HASensorNumber Temp6("Temp6", HASensorNumber::PrecisionP1);
HASensorNumber Temp7("Temp7", HASensorNumber::PrecisionP1);
HASensorNumber Temp8("Temp8", HASensorNumber::PrecisionP1);

HASensorNumber Volt1("Volt1", HASensorNumber::PrecisionP2);
HASensorNumber Volt2("Volt2", HASensorNumber::PrecisionP2);
HASensorNumber Volt3("Volt3", HASensorNumber::PrecisionP2);
HASensorNumber Volt4("Volt4", HASensorNumber::PrecisionP2);
HASensorNumber Volt5("Volt5", HASensorNumber::PrecisionP2);
HASensorNumber Volt6("Volt6", HASensorNumber::PrecisionP2);
HASensorNumber Volt7("Volt7", HASensorNumber::PrecisionP2);
HASensorNumber Volt8("Volt8", HASensorNumber::PrecisionP2);

I attached my init code.

We played around with HA and defined the sensors manually, we get some data, but not consistent and reliable.
BTW, HA is running stable on a local machine and the whole house is implemented (KNX and other devices).

Best, Christof

Regards
Freundliche Grüsse

Christof Schmid

AAA Immobilien AG
Seestrasse 81b│8700 Küsnacht │Tel. 043 255 00 00

Telefon: 043 255 00 00
Mobile: 079 688 06 60

ChristofSchmid, 2023-07-02:
Hi Emil
This did the trick.

HAMqtt mqtt(client, device, MAXSENSORS);

Cheers, Christof

EmilEmilchen, 2023-07-05:
Hi again, maybe this isnt the right place for this but does anyone - by any chance - have any idea what I could use as an alternative to mqtt in general or at least to this library in particular (in case I dont get it working) to be able to communicate with homeassistant. Here it would be really nice to have something that does state reporting   to HA. I thought about using just http or tcp/udp but I would like to have something better suitable for iot things. And I looked into ESP home and some solutions to use a serial connection to the arduino but its not really possible for me to have a direct physical USB/Serial connetion between the Arduino and my HA Server. Kind regards, Emil

ChristofSchmid, 2023-07-06:
Hi Emil
I am not sure if i understand your code right.
But is the line 
HASwitch* led[2];
correct to initialize for 8 relay switches?
In your sketch you refer to more than 2 leds.

EmilEmilchen, 2023-07-06:
Now I'm dont even understand my own code anymore... But I'm pretty sure that the line is there just by mistake. I't shouldnt interfer with anything though since at this point led = [], so led[] should be undefined. And yes, I do create 8 switches. But only in the for loop in void setup(), I loop through the arrays defined at the top in order to create all the HASwitches, and all of this works flawlessly with 4 or even 5 switches. But as soon as I have 6 more some just dont work and shows weird behavior as described in my initial posting.

ChristofSchmid, 2023-07-06:
IMHO it should read "HASwitch* led[pin_count]". Maybe this causes a wild read in unknown memory areas with unpredictable results.

EmilEmilchen, 2023-07-14:
Thanks for the advice but turns out: no. That's not the problem, just removing the line in question doesnt cause any change in behavior whatsoever, neither does changing it to `HASwitch* led[pin_count]`.

Tonguc-Endem, 2023-08-26:
> so, kids are sleeping; had a look at the code; number of devices is limited to 6 by default
>
> from HAMqtt.h ` explicit HAMqtt( Client& netClient, HADevice& device, const uint8_t maxDevicesTypesNb = 6 [...]

Your comment helped me so much.  I thought it was a bug and I wasted so many hours trying to find a workaround.
Thank you so much 👍

closed 2024-01-30 by dawidchyrzynski