dawidchyrzynski/arduino-home-assistant #182: [Feature Request] Add configuration_url to Device configuration

MrRodz, 2023-08-05:
Hi Would it be possible to add configuration_url to the Device configuration section to allow the visit button on home assistant. Thanks

erkr, 2023-11-26:
Yes please, I want to link the site with firmware updates 😊

One note: before adding the URL, optimise the publishConfig() first. 
Now every device will publish the full device context. That is only needed for one device. All others only need to include the identifier. That will reduce the overall config payload, especially when adding an URL.

closed 2024-02-05 by dawidchyrzynski

muyan3030, 2024-11-17:
char configUrl[64];
  String localIp = WiFi.localIP().toString();
  snprintf(configUrl, sizeof(configUrl), "http://%s", localIp.c_str());
  Serial.println(configUrl);
  device.setConfigurationUrl(configUrl);
  
  http://192.168.5.112
  
  I'm trying to use setConfigurationUrl but getting these logs, can anyone help me?
  
WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"name":"Pretty label 1","uniq_id":"e8db84dab214_mySwitch1","ic":"mdi:lightbulb","dev":{"ids":"e8db84dab214","name":"ESP_LIGHT_DAB214","mf":"DIY","mdl":"ESP8266","sw":"1.50.20241116.HW20","cu":"\xff\xff\xff\xff\x08\r\xff?\xdc\x0c\xff?\x02\r#@\xff\xff\xff\xff\x08\r\xff?\x01"},"stat_t":"aha/e8db84dab214/mySwitch1/stat_t","cmd_t":"aha/e8db84dab214/mySwitch1/cmd_t"}' on homeassistant/switch/e8db84dab214/mySwitch1/config with encoding utf-8 (for <Job None HassJobType.Callback <function async_start.<locals>.async_discovery_message_received at 0x7fb8de338d60>>)