dawidchyrzynski/arduino-home-assistant #183: sensor uniqueId vs device uniqueId vs entity_id in home assistant

mo22, 2023-08-08:
Hi,

it looks as if the home assistant mqtt integration only uses the unique id of the sensor for entity_id naming - so having two instances both with a number sensor with unique id "temperature" they map both to the same entity id in home assistant.

I believe the unique ID in the discovery data generated per sensor should be prefixed with the unique ID of the device.

daniloc, 2023-08-14:
I had the same issue. Funny enough, v1.3.0 worked exactly this way:

https://github.com/dawidchyrzynski/arduino-home-assistant/blob/a4e50b7d048373d1254fef2e3e48f5e9a5fd51fc/src/device-types/DeviceTypeSerializer.cpp#L228C1-L250C2

I've created a PR that restores this functionality by appending the MQTT device ID to all sub-devices upon construction:

https://github.com/dawidchyrzynski/arduino-home-assistant/pull/186

mano3m, 2023-10-08:
struggling with this one as well. I made serveral devices with the same sensors and can only get one to work unless I hardcode different IDs to all of them.

daniloc, 2023-10-08:
@mano3m If you're using PlatformIO, you can swap in this URL in `platformio.ini` under `lib_deps` and be right on your way until this is fixed:

https://github.com/daniloc/arduino-home-assistant.git

dawidchyrzynski, 2024-02-05:
This upcoming feature will be named "Extended Unique IDs" and is set to be released in version 2.1.0 shortly (#217). Feel free to try it on the develop branch.

closed 2024-02-05 by dawidchyrzynski