dawidchyrzynski/arduino-home-assistant #194: Arduino Nano ESP32 - is it support?

BKSnake, 2023-09-19:
I see interesting behaviour my board Arduino Nano 33 IoT connected normaly by example but I rewrited sketch for Arduino Nano ESP32 and it doesn't connect to MQTT Broker.

Here my sketch:
```
#include <WiFi.h>
#include <ArduinoHA.h>

// secrets here
#include "secrets.h"

WiFiClient client;
HADevice device;
HAMqtt mqtt(client, device);

// "led" is unique ID of the switch. You should define your own ID.
HASwitch led("led");

void onSwitchCommand(bool state, HASwitch* sender) {
  digitalWrite(LED_BUILTIN, (state ? HIGH : LOW));
  sender->setState(state);  // report state back to the Home Assistant
}

void onMqttConnected() {
  Serial.println("Connected to the broker!");
}

void setup() {
  Serial.begin(9600);
  Serial.println("Starting...");

  // Unique ID must be set!
  byte mac[6];
  WiFi.macAddress(mac);
  device.setUniqueId(mac, sizeof(mac));

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  // connect to wifi
  WiFi.begin(SSID, PASS);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);  // waiting for the connection
  }
  Serial.println();
  Serial.println("Connected to the network");

  // set device's details (optional)
  device.setName("Test Sensor");
  device.setModel("Arduino Nano ESP32");
  device.setSoftwareVersion("0.0.1");

  // handle switch state
  led.onCommand(onSwitchCommand);
  led.setName("Build State");  // optional

  mqtt.onConnected(onMqttConnected);
  mqtt.begin(BROKER_ADDR, BROKER_USER, BROKER_PASS);
}

void loop() {
  mqtt.loop();

  // You can also change the state at runtime as shown below.
  // This kind of logic can be used if you want to control your switch using a button connected to the device.
  // led.setState(true); // use any state you want
}

Could you help me to understand problem?
Thanks

wanderingwonderer, 2023-09-29:
You don’t assign any value to mac?

BKSnake, 2023-09-30:
I found problem - 

```HASwitch led("led");``` by documentation it should be unique ))))

I have last questions. 

how can i send correctly battery level to Home Assistant?

thanks

wanderingwonderer, 2023-09-30:
I’m sorry, I don’t have the answer to that. You should probably close this issue and ask that question in Discussions.

BKSnake, 2023-10-01:
Resolved problems

Thanks

closed 2023-10-01 by BKSnake