dawidchyrzynski/arduino-home-assistant #202: _mqtt->setBufferSize

qibao07, 2023-11-03:
Support `setBufferSize`.
```
bool HAMqtt::begin(
    const IPAddress serverIp,
    const uint16_t serverPort,
    const char* username,
    const char* password,
    const uint16_t bufferSize
)
{
    ARDUINOHA_DEBUG_PRINT(F("AHA: init server "))
    ARDUINOHA_DEBUG_PRINT(serverIp)
    ARDUINOHA_DEBUG_PRINT(F(":"))
    ARDUINOHA_DEBUG_PRINTLN(serverPort)

    if (_device.getUniqueId() == nullptr) {
        ARDUINOHA_DEBUG_PRINTLN(F("AHA: init failed. Missing device unique ID"))
        return false;
    }

    if (_initialized) {
        ARDUINOHA_DEBUG_PRINTLN(F("AHA: already initialized"))
        return false;
    }

    _username = username;
    _password = password;
    _initialized = true;

    _mqtt->setServer(serverIp, serverPort);
    _mqtt->setCallback(onMessageReceived);
    _mqtt->setBufferSize(bufferSize);

    return true;
}
```

dawidchyrzynski, 2024-01-23:
Hi @qibao07, what's the use case?

qibao07, 2024-01-24:
@dawidchyrzynski 
Thank you for your reply. Sometimes the received message is too long, causing the message to be truncated. Modifying this value can only modify imported code files.

dawidchyrzynski, 2024-01-30:
@qibao07 Do you mean the messages that the library handles/generates, or are you hadling your custom messages?

qibao07, 2024-01-31:
@dawidchyrzynski I implemented HAText myself with reference to HANumber. It was found that long messages require large buffer. Custom messages are also required.

dawidchyrzynski, 2024-02-09:
This feature will be available in the upcoming version of the library. Please refer to the linked Pull Request for more information.

closed 2024-02-09 by dawidchyrzynski