dawidchyrzynski/arduino-home-assistant #204: [Fix suggestion] Max Number of sensors (devices-types)

deckard-rick, 2023-11-12:
Problem:
I have one device with 10 HANumber sensors but config-messages are only send for the first 5.

Background:
I believe, if the HANumbers are created, via pubslihConfig (protected), the config-message is put into a queue. mqtt is not connected if this happens. It looks like, that the queue has a maximum length of five, because of memory size restriction.

Solution:
* HADeviceTypes::publishConfig is now public, instead of protected, in my code.
* I send one time a publishConfig Message before I set my first value to the sensor. Between each of this messages mqtt.loop() is called.
* Work's perfectly

mariansoban, 2023-11-29:
I have same issue for HASwitch - there are 6 HASwitch instances created in code, but there only 5 switches synchronized to Home Assistant

yannicksuter, 2024-01-16:
@deckard-rick override the max number of devices allowed to be registered in the HAMqtt constructor and your problem is solved. I ran into exactly the same issue.

dawidchyrzynski, 2024-01-23:
https://dawidchyrzynski.github.io/arduino-home-assistant/documents/library/device-types.html