dawidchyrzynski/arduino-home-assistant #205: Can somebody please send me code that I can just copy paste into the Arduino IDE text box and press the Upload button?

Slyceth, 2023-11-17:
I have zero experience with anything programming, and the documentation for this project gives me a headache, there's pages upon pages with chunks of code that I have no clue what I have to use and where.

All I need is my arduino UNO R4 WiFi to send a signal that triggers a script called "LIGHTPRESS" whenever I point a laserpen at my arduino. (Lightpress is a script that I can then put my lights Toggle in)

I found this code online that works perfectly, but instead of making a beep sound I need it to trigger an event of my choice inside Home Assistant:

`int ldrPin = A4;
int sirenPin = 11;
int ledPin = A0;
long ldrValue1, ldrValue2;

void setup(void) {
  pinMode (sirenPin,OUTPUT);
  pinMode (ledPin,OUTPUT);
  pinMode (ldrPin,INPUT);
  //Serial.begin(9600);
}

void loop(void) {
  ldrValue1 = analogRead(ldrPin);
  delay(10);
  ldrValue2 = analogRead(ldrPin);

  if (ldrValue1-ldrValue2 > 20||ldrValue2-ldrValue1 > 20){
    digitalWrite(sirenPin,HIGH);
    digitalWrite(ledPin,HIGH);
    delay(1000);
  }
  else{
    digitalWrite(sirenPin,LOW);
    digitalWrite(ledPin,LOW);
  }
}`

Please help me out, I am in the programming hole where I know exactly what I need it to do but no idea how to code it.

closed 2024-01-08 by dawidchyrzynski

Slyceth, 2024-01-09:
> <dawidchyrzynski/arduino-home-assistant/issue/205/issue_event/11422975395@
> github.com>

I don't think it's completed yet? I still have major issues with this

dawidchyrzynski, 2024-01-09:
Hi @Slyceth,

This isn't a bug report or a feature request. I'm unable to provide support for your project. Please refer to the examples in the repository or consider posting your issue on the Arduino forum for assistance.