dawidchyrzynski/arduino-home-assistant #209: Entity not returning online in Home Assistant after power cutoff

clemente3905, 2023-12-12:
Hi, I have a device that use an arduino to create a switch and 2 number sensor to home assistant.

After first connection everything was fine.

Then I had a power cutoff and when power come back to the arduino, home assistant was unable to see it available again.

How can I make this problem away?

Following is the code in arduino sketch (full code)

`
#include <WiFiNINA.h>
#include <ArduinoHA.h>

char ssid[]= SECRET_SSID;
char pass[] = SECRET_PASS;
const char broker[] = SECRET_MY_BROKER_IP;
const char mqtt_user[] = SECRET_MY_BROKER_USER;
const char mqtt_pass[] = SECRET_MY_BROKER_PASS;
const int port = SECRET_MY_BROKER_PORT;

#define sw1 11 //pin 11 di comando del relè1
#define sw2 12 //pin 12 di comando del relè2
#define SPENTO HIGH //funzionano in logica inversa
#define ACCESO LOW  //funzionano in logica inversa
#define LO_Threshold 500 //DA REGOLARE, soglia di apertura carico acqua
#define HI_Threshold 800 //DA REGOLARE, soglia di chiusura carico acqua
#define SENSORE A4 //DA CORREGGERE, mettere il sensore di pressione
unsigned long lastUpdateAt = 0;
const unsigned int sogliaControllo = 5000; //controllo il dato ogni 5 secondi

WiFiClient client;
HADevice device("Ard-CaricoH2O");
HAMqtt mqtt(client, device);
HASwitch elettrovalvola("elettrovalvola");
HASensorNumber analogSensor("myAnalogInput", HASensorNumber::PrecisionP1);
HASensorNumber analogSensor2("letturaDigitale");

void onSwitchCommand(bool state, HASwitch* sender)
{
    digitalWrite(sw1, (state ? ACCESO : SPENTO));
    digitalWrite(sw2, (state ? ACCESO : SPENTO));
    sender->setState(state); // report state back to the Home Assistant
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  while (!Serial) {
    ; //aspetto di connettermi alla seriale
  }
    
  device.enableSharedAvailability();
  device.enableLastWill();
  device.setName("Arduino Carico Automatico");
  device.setSoftwareVersion("1.0.0");
  device.setManufacturer("DoItYourself Corp.");
  device.setModel("Arduino Uno WiFi Rev.2");
  elettrovalvola.setName("Elettrovalvola");
  analogSensor.setIcon("mdi:water-pump");
  analogSensor.setName("Pressione Acqua");
  analogSensor.setUnitOfMeasurement("atm");
  analogSensor2.setIcon("mdi:numeric");
  analogSensor2.setName("Lettura Digitale");
  analogSensor2.setUnitOfMeasurement("(0->1023)");

  pinMode(A5, INPUT);
  pinMode(A4, INPUT);
  pinMode(sw1, OUTPUT);
  pinMode(sw2, OUTPUT);
  digitalWrite(sw1, SPENTO);
  digitalWrite(sw2, SPENTO);
  Serial.print("Connessione in corso alla WiFi ");
  Serial.println(ssid);
  while (WiFi.begin(ssid,pass) != WL_CONNECTED) {
    Serial.print(".");
    delay(5000);
  }

  //qui sono connesso alla WiFi
  Serial.println();
  Serial.println("WiFi CONNESSA.");
  Serial.println();

  //digitalWrite(sw1, SPENTO);
  //digitalWrite(sw2, SPENTO);
  elettrovalvola.onCommand(onSwitchCommand);

  Serial.println("Connessione in corso al broker ");
  Serial.println(broker);
  mqtt.begin(broker, port, mqtt_user, mqtt_pass);
 
}

void loop() {
  // put your main code here, to run repeatedly:
  mqtt.loop();

  if ((millis() - lastUpdateAt) > sogliaControllo) {
    uint16_t reading = analogRead(SENSORE);
    float lettura = reading * 5.f / 1023.f; //0.0 - 5.0 
    Serial.print("\nLettura analogica pari a ");
    Serial.println(lettura);
    Serial.print("Lettura digitale pari a ");
    Serial.println(reading);

    analogSensor.setValue(lettura);
    analogSensor2.setValue(reading);
    lastUpdateAt = millis();

    //controllo automatico
    if (reading < LO_Threshold) {
      digitalWrite(sw1, ACCESO);
      digitalWrite(sw2, ACCESO);
      elettrovalvola.setState(true);
    }

    if (reading > HI_Threshold) {
      digitalWrite(sw1, SPENTO);
      digitalWrite(sw2, SPENTO);
      elettrovalvola.setState(false);
    }

  }

}
`

clemente3905, 2023-12-12:
Just to complete: if I connect to arduino with laptop I can see via serial monitor that it is working as expected (wifi connected, analog read correctly performed, automatic action correctly performed).
But it is offline in home assistant and will not come back online...

klausj1, 2023-12-30:
Did you check if the values arrive in the MQTT broker (e.g. with MQTT explorer)? If yes, the problem is on HA side.