dawidchyrzynski/arduino-home-assistant #214: Is the number of HASensorNumber objects/values reported to MQTT broker limited?

yannicksuter, 2024-01-15:
I'm fighting a strange effect, that only the values of the first 5 HASensorNumber objects are pushed to the MQTT broker, even thought I have more defined. I commented the first 5 sensor objects out and then the next 5 appear - this is I verified that I had no error in setting up the HASensorNumber instances. 

Is there a hard limit how many "metrics" can be tracked/published per haDevice? can this be changed somehow?

yannicksuter, 2024-01-16:
I checked with MQTT Explorer what get published to HA (Mosquitto) and found the following situation -> all sensors are in aha -> but only 5 of them get also published under the homeassistant topic.. therefore not appearing in HA. What could cause this?

[image]

morganzion, 2024-01-16:
Make sure to check other issues as well. Some PR have not been integrated by the repo owner, in fact this repo seems dead :/