dawidchyrzynski/arduino-home-assistant #225: Fan does not work

LuckyG3000, 2024-02-07:
Hi Dawid,
I'm a new user of HA and your library. I have a couple of devices in my home that I made with my own firmware, they are using ATmega 644P, 1284P, 2560 with W5200 or W5500, and of course ESP8266. Now I'm trying to integrate your library into these devices. After initial issues with not enough free memory, I finally got it working with simple LED/switch control. But what does not work for me is the fan using HAFan. After investigating I think the control properties for Fan in HA changed in past and the current version of HA does not work with "speed" anymore (and "spd_rng_min" / "spd_rng_max") but instead uses [percentage and presets](https://www.home-assistant.io/integrations/fan/).
Is there any chance you could update the library so it would work with current Fan implementation in HA? Or am I not understanding the way it works?

closed 2024-02-09 by LuckyG3000

dawidchyrzynski, 2024-02-09:
Hi @LuckyG3000, ensure that you are using the most recent version (2.0.0) of the library, as it matches to the latest implementation of the Fan in Home Assistant. Can you confirm that you have the latest version installed?

LuckyG3000, 2024-02-09:
> Hi @LuckyG3000, ensure that you are using the most recent version (2.0.0) of the library, as it matches to the latest implementation of the Fan in Home Assistant. Can you confirm that you have the latest version installed?

Hi Dawid,
Yes, I was using the recent version 2.0.0. Later I found out in HA logs what might have been the problem:

`[homeassistant.components.mqtt.mixins] Error 'speed_range_min must be > 0'`

Yes, I was setting the `speed_range_min` to zero. I apologize for my mistake.
Anyway after I finally understood the whole MQTT discovery and the message format it accepts, I moved to sending the messages directly via PubSubClient library. I prefer to not use char arrays for such long strings but rather sending them from code memory with beginPublish(), write() and endPublish() methods. It saves a lot of RAM on ATmega644P.
Thank you for reaching out!

dawidchyrzynski, 2024-02-09:
@LuckyG3000 The library internally streams the data as you described (beginPublish / write / endPublish). The largest buffer used is below 64 bytes and the library itself doesn't allocate anything else than PubSubClient buffer and device types storage (< 24 bytes). It was optimized to utilize the lowest RAM possible.